verilator/verilator #4909: Enhance Verilator to include compile and runtime statistics in its output at the End

opensource-elearning, 2024-02-20:
Thanks for taking the time to report this.

**What would you like added/supported?**

Enhance Verilator to include compile and runtime statistics in its output:

- Compilation Time: Report the CPU time taken to compile the Verilog/SystemVerilog source code.
- Elaboration Time: Report the CPU time taken to elaborate the design and generate the circuit model.
- Link Time (if applicable): Report the time taken to link the design with any external libraries or components.
- Simulation Time: Report the CPU time taken to simulate the design.

This information would provide valuable insights into the performance of different hardware configurations and designs, without the need for additional debugging flags. It would be displayed as 5 additional lines in the terminal output or logs, providing a concise summary of the compile and runtime statistics.

**Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)**

```
           V e r i l a t o r   S i m u l a t i o n   R e p o r t 

<Verilator Version information>
CPU Time
Compilation Time:   9.935 seconds
Elaboration Time:   0.288 seconds
Linking Time:       0.591 seconds 
Simulation Time:    0.600 seconds

Tue Feb 20 02:08:18 2024 <Date and Time>
```
We may provide the information about the number of processor `proc` used if you feel it would be helpful.

The above report may change for compilation in which we may exclude the Simulation time

**May we assist you in trying to fix this in Verilator yourself?**
No, I don't C/C++

wsnyder, 2024-02-20:
--stats has some of this information.  Runtime reporting is up to the user's main().

opensource-elearning, 2024-02-20:
As many design and verification engineers rely heavily on Verilog/SV-based testbenches and may not utilize C++, it is unclear how the integration of C++ into Verilator can be achieved for many new engineers. However, such an effort would require collaboration from all teams involved in daily Verilator usage.

To enhance user experience, it is suggested that basic statistics be printed at the end of the simulation or compilation without the need for additional arguments or flags like `--stats`. While the `--stats` flag provides valuable debugging information, it can be overwhelming for routine tasks.

A simple summary of relevant information at the end of simulations or compilations would be beneficial to design and verification teams that rely on Verilator for daily basis. The implementation of this feature is at the discretion of the development team.

If the effort required is minimal, it would be greatly appreciated if this request could be considered.

wsnyder, 2024-02-21:
Created a formal survey here: https://github.com/verilator/verilator-announce/issues/67

wsnyder, 2024-03-01:
Some proposed examples for discussion.  I want to keep the line count small, width < 100 characters, have a leading "-" to make filtering out easy.

```
- Verilator 5.123 2024-01-01: B u i l d   R e p o r t
- Verilator: Built from 235 KB sources in 23 modules, into 123 KB in 23 C++ files needing 123MB
- Verilator: Walltime 1234.1 s (elab=1.1, cvt=4.2, comp=6.3); cpu 22 s on 4 threads; alloced 123MB

- Verilator 5.123 2024-01-01: S i m u l a t i o n   R e p o r t
- Verilator: End at simtime 123 ns: walltime 1234.001 s; speed 123.1 ns/s
- Verilator: cpu 22.001 s on 4 threads; alloced 123 MB
```

Build information
- "Verilator 5.123 2024-01-01" - version
- "235 KB sources" - characters of post-preprocessed non-whitespace text in all inputted Verilog files.
- "23 modules" - number of interfaces/modules/classes/packages
- "into 123 KB" - characters of output non-whitespace C++ code
- "23 C++ files" - number of .cpp files created
- "needing 123MB" = build-time estimate of memory needed to run model
- "Walltime 1234.1 s" - seconds of real elapsed wall time
- "elab=1.1" - seconds to read in files and complete elaboration
- "cvt=4.2" - seconds for verilator to process and write output
- "comp=6.3" - seconds to compile gcc/clang (if using --build)
- "cpu 22 s" - CPU time used total across all CPU threads
- "4 threads" - number of simultaneous threads used
- "alloced 123MB" - total memory used during build

Run information
- "Verilator 5.123 2024-01-01" - version
- "End at simtime 123 ns" - Verilog $time when exiting
- "walltime 1234.001 s" - seconds of real elapsed wall time
- "speed 123.1 ns/s" - simulated time (if non-zero) divided by wall time excluding initialization.  e.g. for a 1 GHz clock, 123 ns/s means 123 cycles per second.  The units are automatically selected to give a number between 1 and 1000.
- "cpu 22 s" - CPU time used total across all CPU threads
- "4 threads" - number of simultaneous threads used
- "alloced 123MB" - total memory used during build

Flags and other notes
- `--quiet` - disable build time statistics printing
- `+verilator+quiet` - disable run time statistics printing
- Will add a VerilatorContext method to print the information, so user's main() may use it.

opensource-elearning, 2024-03-01:
Thanks @wsnyder for accepting this feature request.
I believe that having at least 2 lines statistics information would be very helpful in daily use.
I am happy with the information provided, but I would like to invite contributors/development teams to provide input, suggestions, or feedback if they have any. @gezalore , @RRozak , @yTakatsukasa , @imphil , @kiryk , @kbieganski , @wallento and all developers

kbieganski, 2024-03-01:
> * `--quiet` - disable build time statistics printing

Most people in the survey either want it as an opt-in, or don't want it/care about it. So maybe it'd be better to have a `--verbose`? Though I guess it may be confusing to have it work the other way for simulation.

opensource-elearning, 2024-03-04:
We are aligned with your proposed idea and flags `--quite` and `+quite` as per majority.

You may merge your changes.

poemonsense, 2024-04-10:
> > * `--quiet` - disable build time statistics printing
> 
> Most people in the survey either want it as an opt-in, or don't want it/care about it. So maybe it'd be better to have a `--verbose`? Though I guess it may be confusing to have it work the other way for simulation.

I have this question as well. Most people from the survey don't want it at all. Why would Verilator enable it by default? After I bump v5.024, the summary is really annoying.

PR #5018 (2024-03-25): Add printing summary reports (#4909).
Adds Verilator and Runtime summary reports, controlled by --quiet and +verilator+quiet.

Closes #4909.