verilator/verilator #4920: Transitively parameterized module gives incorrect width-trunc error

JosseVanDelm, 2024-02-23:
Hi!

We are developing a course at our uni and we are switching over to verilator, 
from previously running on commercial tools.
I believe I encountered a case in which verilator is not working as expected.

This issue might be related to https://github.com/verilator/verilator/issues/4529 , but I'm not sure

I'm putting everything between a `<details>` because students are not really supposed to know the solution to this exercise :crossed_fingers: :hand_over_mouth: 

This RTL (alu.sv):

<details>

```systemverilog
module mux2 #(parameter int N = 4)
  (input [N-1:0] d0, d1,
   input s,
   output [N-1:0] y);

  assign y = s ? d1 : d0;
endmodule

module mux4 #(parameter int N = 4)
  (input [N-1:0] d0, d1, d2, d3,
   input [1:0] S,
   output [N-1:0] y);

  wire [N-1:0] output_low, output_high;

  mux2 #(N) lowmux( .d0(d0), .d1(d1), .s(S[0]), .y(output_low));
  mux2 #(N) highmux( .d0(d2), .d1(d3), .s(S[0]), .y(output_high));
  mux2 #(N) finalmux( .d0(output_low), .d1(output_high), .s(S[1]), .y(y));

endmodule
```
</details>

With this testbench (tb_mux4.sv) :

<details>

```systemverilog
`include "alu.sv"

// Code your testbench here
// or browse Examples
module tb_mux4 ();
  // mux4 tests
  reg [5:0] d0, d1, d2, d3;
  reg  [1:0] S;
  wire [5:0] Y;

  mux4 #(6) dut(.d0(d0), .d1(d1), .d2(d2), .d3(d3), .S(S), .y(Y));

  initial begin
    // mux4 tests
        d0 = 6'b000000; d1 = 6'b000001; d2 = 6'b000010; d3 = 6'b000100;
        S = 2'b00; #10;
        S = 2'b01; #10;
  end
endmodule
```

</details>

compiles just fine with
```
verilator --binary --timing --trace --assert -j 0 -Wno-MULTITOP tb_mux4.sv
```
However, if I add the rest of the alu.sv file:

<details>

```systemverilog
// Code your design here

module mux2 #(parameter int N = 4)
  (input [N-1:0] d0, d1,
   input s,
   output [N-1:0] y);

  assign y = s ? d1 : d0;
endmodule

module mux4 #(parameter int N = 4)
  (input [N-1:0] d0, d1, d2, d3,
   input [1:0] S,
   output [N-1:0] y);

  wire [N-1:0] output_low, output_high;

  mux2 #(N) lowmux( .d0(d0), .d1(d1), .s(S[0]), .y(output_low));
  mux2 #(N) highmux( .d0(d2), .d1(d3), .s(S[0]), .y(output_high));
  mux2 #(N) finalmux( .d0(output_low), .d1(output_high), .s(S[1]), .y(y));

endmodule

module mux8 #(parameter int N = 4) // maybe implement this differently
  (input [N-1:0] d0, d1, d2, d3, d4, d5, d6, d7,
   input [2:0] S,
   output [N-1:0] y);

  wire [N-1:0] output_low, output_high;

  mux4 #(N) lowmux(.d0(d0), .d1(d1), .d2(d2), .d3(d3), .S(S[1:0]), .y(output_low));
  mux4 #(N) highmux(.d0(d4), .d1(d5), .d2(d6), .d3(d7), .S(S[1:0]), .y(output_high));
  mux2 #(N) finalmux(.d0(output_low), .d1(output_high), .s(S[2]), .y(y));

endmodule
```

</details>

Verilator seems to not be able to see that the parameters in mux2 are "transitively" set in mux4?

```
verilator --binary --timing --trace --assert -j 0 -Wno-MULTITOP tb_mux4.sv
%Warning-WIDTHTRUNC: alu.sv:18:22: Input port connection 'd0' expects 4 bits on the pin connection, but pin connection's VARREF 'd0' generates 6 bits.
                                 : ... note: In instance 'tb_mux4.dut'
   18 |   mux2 #(N) lowmux( .d0(d0), .d1(d1), .s(S[0]), .y(output_low));
      |                      ^~
                     tb_mux4.sv:2:1: ... note: In file included from 'tb_mux4.sv'
                     ... For warning description see https://verilator.org/warn/WIDTHTRUNC?v=5.020
                     ... Use "/* verilator lint_off WIDTHTRUNC */" and lint_on around source to disable this message.
%Warning-WIDTHTRUNC: alu.sv:18:31: Input port connection 'd1' expects 4 bits on the pin connection, but pin connection's VARREF 'd1' generates 6 bits.
                                 : ... note: In instance 'tb_mux4.dut'
   18 |   mux2 #(N) lowmux( .d0(d0), .d1(d1), .s(S[0]), .y(output_low));
      |                               ^~
                     tb_mux4.sv:2:1: ... note: In file included from 'tb_mux4.sv'
%Warning-WIDTHTRUNC: alu.sv:18:50: Output port connection 'y' expects 4 bits on the pin connection, but pin connection's VARREF 'output_low' generates 6 bits.
                                 : ... note: In instance 'tb_mux4.dut'
   18 |   mux2 #(N) lowmux( .d0(d0), .d1(d1), .s(S[0]), .y(output_low));
      |                                                  ^
                     tb_mux4.sv:2:1: ... note: In file included from 'tb_mux4.sv'
%Warning-WIDTHTRUNC: alu.sv:19:23: Input port connection 'd0' expects 4 bits on the pin connection, but pin connection's VARREF 'd2' generates 6 bits.
                                 : ... note: In instance 'tb_mux4.dut'
   19 |   mux2 #(N) highmux( .d0(d2), .d1(d3), .s(S[0]), .y(output_high));
      |                       ^~
                     tb_mux4.sv:2:1: ... note: In file included from 'tb_mux4.sv'
%Warning-WIDTHTRUNC: alu.sv:19:32: Input port connection 'd1' expects 4 bits on the pin connection, but pin connection's VARREF 'd3' generates 6 bits.
                                 : ... note: In instance 'tb_mux4.dut'
   19 |   mux2 #(N) highmux( .d0(d2), .d1(d3), .s(S[0]), .y(output_high));
      |                                ^~
                     tb_mux4.sv:2:1: ... note: In file included from 'tb_mux4.sv'
%Warning-WIDTHTRUNC: alu.sv:19:51: Output port connection 'y' expects 4 bits on the pin connection, but pin connection's VARREF 'output_high' generates 6 bits.
                                 : ... note: In instance 'tb_mux4.dut'
   19 |   mux2 #(N) highmux( .d0(d2), .d1(d3), .s(S[0]), .y(output_high));
      |                                                   ^
                     tb_mux4.sv:2:1: ... note: In file included from 'tb_mux4.sv'
%Warning-WIDTHTRUNC: alu.sv:20:24: Input port connection 'd0' expects 4 bits on the pin connection, but pin connection's VARREF 'output_low' generates 6 bits.
                                 : ... note: In instance 'tb_mux4.dut'
   20 |   mux2 #(N) finalmux( .d0(output_low), .d1(output_high), .s(S[1]), .y(y));
      |                        ^~
                     tb_mux4.sv:2:1: ... note: In file included from 'tb_mux4.sv'
%Warning-WIDTHTRUNC: alu.sv:20:41: Input port connection 'd1' expects 4 bits on the pin connection, but pin connection's VARREF 'output_high' generates 6 bits.
                                 : ... note: In instance 'tb_mux4.dut'
   20 |   mux2 #(N) finalmux( .d0(output_low), .d1(output_high), .s(S[1]), .y(y));
      |                                         ^~
                     tb_mux4.sv:2:1: ... note: In file included from 'tb_mux4.sv'
%Warning-WIDTHTRUNC: alu.sv:20:69: Output port connection 'y' expects 4 bits on the pin connection, but pin connection's VARREF 'y' generates 6 bits.
                                 : ... note: In instance 'tb_mux4.dut'
   20 |   mux2 #(N) finalmux( .d0(output_low), .d1(output_high), .s(S[1]), .y(y));
      |                                                                     ^
                     tb_mux4.sv:2:1: ... note: In file included from 'tb_mux4.sv'
%Error: Exiting due to 9 warning(s)
make: *** [../Makefile:45: obj_dir/Vtb_mux4] Error 1

```

If I set both parameters to 6 (see below) it compiles just fine again.

<details>

```systemverilog
// Code your design here

module mux2 #(parameter int N = 6)
  (input [N-1:0] d0, d1,
   input s,
   output [N-1:0] y);

  assign y = s ? d1 : d0;
endmodule

module mux4 #(parameter int N = 6)
  (input [N-1:0] d0, d1, d2, d3,
   input [1:0] S,
   output [N-1:0] y);

  wire [N-1:0] output_low, output_high;

  mux2 #(N) lowmux( .d0(d0), .d1(d1), .s(S[0]), .y(output_low));
  mux2 #(N) highmux( .d0(d2), .d1(d3), .s(S[0]), .y(output_high));
  mux2 #(N) finalmux( .d0(output_low), .d1(output_high), .s(S[1]), .y(y));

endmodule

module mux8 #(parameter int N = 4) // maybe implement this differently
  (input [N-1:0] d0, d1, d2, d3, d4, d5, d6, d7,
   input [2:0] S,
   output [N-1:0] y);

  wire [N-1:0] output_low, output_high;

  mux4 #(N) lowmux(.d0(d0), .d1(d1), .d2(d2), .d3(d3), .S(S[1:0]), .y(output_low));
  mux4 #(N) highmux(.d0(d4), .d1(d5), .d2(d6), .d3(d7), .S(S[1:0]), .y(output_high));
  mux2 #(N) finalmux(.d0(output_low), .d1(output_high), .s(S[2]), .y(y));

endmodule
```

</details>

> What 'verilator --version' are you using?  Did you try it with the git master version?
```  
$ verilator --version
Verilator 5.020 2024-01-01 rev UNKNOWN.REV
```
> What OS and distribution are you using?

I'm running this on Arch Linux with kernel 6.6.11-1-lts

> May we assist you in trying to fix this in Verilator yourself?

I don't think I have the skills to do that :sweat_smile:

Is this in fact unexpected behaviour? Thank you so much in advance!

wsnyder, 2024-02-23:
The lowmux is incorrectly getting the default width instead of "N". This is because, partially due to -Wno-MULTITOP, mux4 with the default width is elaborated first and this looses the #(N) on lowmux. Then when followed by an elaboration of mux4 #(6) the lowmux parameter is missing.
Fix is to save modules during elaboration similar to how classes and interfaces are saved.  As usual a "simple" change to do this broke a bunch of tests, so more time is needed to fix this.

@JosseVanDelm note a workaround for your case, is to avoid -Wno-MULTITOP.  Put a celldefine around the student's instantiations in e.g. tb_mux4, and this will ignore any unneeded modules e.g. mux8.

<details>

```systemverilog
`celldefine
`include "alu.sv"
`endcelldefine

...
```
</details>

 BTW I suspect google will still index the details; if you want to edit your example to remove the info so student's can't find it in a google search, I've copied it.

JosseVanDelm, 2024-02-26:
@wsnyder okay thanks!
I'll have a look at what fix works best. Putting everything in different files seems like a good practice anyhow.
I also changed the example to something that emits similar behaviour, but which is okay for our students to learn from :)

wsnyder, 2025-09-20:
Fixed earlier in #4497 et al, example passes on master, committing to tests shortly.

closed 2025-09-20 by wsnyder