verilator/verilator #4925: Inout port of unpacked structure type is emitted as char variable

RRozak, 2024-02-26:
I get error when I declare `inout` port of unpacked structure type. It looks that port variables are emitted as variables of type `char`.
Here is an example:
```systemverilog
typedef struct {
   bit x;
} my_struct_t;

module mh1 (inout my_struct_t a, inout my_struct_t b);
   assign b.x = a.x;
endmodule

module t;
   my_struct_t a, b;
   mh1 u_mh1(a, b);
   initial begin
      a.x = 1;
      #1;
      if (b.x) begin
         $write("*-* All Finished *-*\n");
         $finish;
      end
      else begin
         $stop;
      end
   end
endmodule
```
The errors (thrown by gcc) are:
```
In file included from Vunpacked_port__ALL.cpp:5:
Vunpacked_port___024root__DepSet_h4e61aed3__0.cpp: In function ‘VlCoroutine Vunpacked_port___024root___eval_initial__TOP__Vtiming__0(Vunpacked_port___024root*)’:
Vunpacked_port___024root__DepSet_h4e61aed3__0.cpp:23:37: error: request for member ‘__PVT__x’ in ‘vlSelf->Vunpacked_port___024root::t__DOT__a__strong__out3’, which is of non-class type ‘CData’ {aka ‘unsigned char’}
   23 |     vlSelf->t__DOT__a__strong__out3.__PVT__x = 1U;
      |                                     ^~~~~~~~
In file included from /usr/local/share/verilator/include/verilated.h:42,
                 from Vunpacked_port__pch.h:21,
                 from Vunpacked_port.cpp:4,
                 from Vunpacked_port__ALL.cpp:3:
Vunpacked_port___024root__DepSet_h4e61aed3__0.cpp:28:23: error: request for member ‘__PVT__x’ in ‘vlSelf->Vunpacked_port___024root::t__DOT__u_mh1__DOT__b__out__strong__out1’, which is of non-class type ‘CData’ {aka ‘unsigned char’}
   28 |                      .__PVT__x))) {
      |                       ^~~~~~~~
/usr/local/share/verilator/include/verilatedos.h:66:45: note: in definition of macro ‘VL_UNLIKELY’
   66 | # define VL_UNLIKELY(x) __builtin_expect(!!(x), 0)  // Prefer over C++20 [[unlikely]]
      |                                             ^
In file included from Vunpacked_port__ALL.cpp:5:
Vunpacked_port___024root__DepSet_h4e61aed3__0.cpp: In function ‘void Vunpacked_port___024root___act_sequent__TOP__0(Vunpacked_port___024root*)’:
Vunpacked_port___024root__DepSet_h4e61aed3__0.cpp:41:54: error: request for member ‘__PVT__x’ in ‘vlSelf->Vunpacked_port___024root::t__DOT__u_mh1__DOT__b__out__strong__out1’, which is of non-class type ‘CData’ {aka ‘unsigned char’}
   41 |     vlSelf->t__DOT__u_mh1__DOT__b__out__strong__out1.__PVT__x
      |                                                      ^~~~~~~~
Vunpacked_port___024root__DepSet_h4e61aed3__0.cpp:42:49: error: request for member ‘__PVT__x’ in ‘vlSelf->Vunpacked_port___024root::t__DOT__a__strong__out3’, which is of non-class type ‘CData’ {aka ‘unsigned char’}
   42 |         = (1U & vlSelf->t__DOT__a__strong__out3.__PVT__x);
      |                                                 ^~~~~~~~
```

wsnyder, 2024-02-26:
Here's an updated test using packed that shows the problem is more severe than just unpacked.  Is this something you'll be looking into fixing?

```
// DESCRIPTION: Verilator: Verilog Test module
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2024 by Ryszard Rozak.
// SPDX-License-Identifier: CC0-1.0

typedef struct packed {
   bit x;
} p_struct_t;

typedef struct {
   bit x;
} u_struct_t;

module p_mh (inout p_struct_t p_i, inout p_struct_t p_o);
   // OK: module p_mh (input p_struct_t p_i, output p_struct_t p_o);
   assign p_o.x = p_i.x;
endmodule

module u_mh (inout u_struct_t u_i, inout u_struct_t u_o);
   // OK: module u_mh (input u_struct_t u_i, output u_struct_t u_o);
   assign u_o.x = u_i.x;
endmodule

module t;
   p_struct_t p_i, p_o;
   u_struct_t u_i, u_o;

   p_mh p_mh(p_i, p_o);
   u_mh u_mh(u_i, u_o);

   initial begin
      p_i.x = 1;
      u_i.x = 1;
      #1;
      if (p_o.x != 1'b1) $stop;
      if (u_o.x != 1'b1) $stop;
      $write("*-* All Finished *-*\n");
      $finish;
   end
endmodule
```

RRozak, 2024-02-26:
Unfortunately I can't work on it (at least in the near future).

RRozak, 2024-03-18:
The problem occurs not only when ports are of struct type.
Here is an example with `inout` ports of `bit` type, which isn't handled correctly too:
```systemverilog
module p_mh (inout bit stu, inout bit xyz);
   assign xyz = stu;
endmodule

module t;
  bit abcd, efgh;

   p_mh p_mh(abcd, efgh);

   initial begin
      abcd = 1;
      #1;
      if (efgh != 1'b1) $stop;
      $write("*-* All Finished *-*\n");
      $finish;
   end
endmodule
```

closed 2024-03-19 by wsnyder