verilator/verilator #4929: When I am using PIO in vscode to generate traces, the report shows success, but the vcd wave file is not generated in the folde

deleteeeee, 2024-02-29:
Installed verialtor on a windos system using Cygwin. gcc-core 11.4 ,verilator v4.228
When I am using PIO in vscode to generate traces, the report shows success, but the vcd wave file is not generated in the folder
[image]

There are only the following files in the .pio folder
[image]

wsnyder, 2024-02-29:
I don't know what PIO is, perhaps whoever makes that framework can help.  If you isolate this to something that only uses Verilator and can post a complete example doing that we'll try to help out.

Note also Cygwin tends to be pretty unstable. If there's any way you can use WSL instead it's much more robust.

deleteeeee, 2024-02-29:
@wsnyder Thanks for the answer. pio refers to the plugin in vscode, platform IO.  Everything looks normal, except that the .vcd file cannot be generated

wsnyder, 2024-03-16:
As above, If you isolate this to something that only uses Verilator and can post a complete example doing that we'll try to help out.

Closing until then.

closed 2024-03-16 by wsnyder