verilator/verilator #4930: Arguments of `$dumpvars` are ignored without a warning

atorregrosa-smd, 2024-02-29:
While building a simple tb with verilator `5.018`, I noticed it ignores the arguments passed to `$dumpvars`, tracing always the whole hierarchy. Calling `tfp->dumpvars()` from the C++ side works as expected.

### How to reproduce

Just use any test compiled with `--binary` and try to restrict the tracing scope with `$dumpvars`. The resulting `vcd` file will show all signals in the design.

wsnyder, 2024-02-29:
I'm reluctant to make this an UNSUPPORTED warning, as the current implementation just "overdelivers", and this is already documented.  But feel free to convince otherwise ;)

https://veripool.org/guide/latest/languages.html?highlight=dumpvars#language-keyword-limitations
> $dumpvars and $dumpports module identifier is ignored; the traced instances will always start at the top of the design. The levels argument is also ignored; use tracing_on/tracing_off pragmas instead.

wsnyder, 2024-03-03:
Closing (for now) per above.

closed 2024-03-03 by wsnyder

atorregrosa-smd, 2024-03-04:
Well, while it's true that it "overdelivers", I would argue that most people calling it with arguments are doing so to limit what is dumped.

I didn't catch the documentation about `$dumpvars` in "Language keyword limitations", I was just looking at https://veripool.org/guide/latest/faq.html?highlight=dumpvars#how-do-i-generate-waveforms-traces-in-c . Maybe a link or comment can be added there?