verilator/verilator #4931: Internal Error: attempted to destroy locked Thread Pool

ollie-etl, 2024-02-29:
I'm seeing flaky failures in ci using verilator 5.020. The failure does not appear to be tied to any specific test, because rerunning typically succeeds. It looks like a race condition in shutdown.

```console 
INFO     cocotb:simulator.py:302 %Error: Internal Error: attempted to destroy locked Thread Pool
INFO     cocotb:simulator.py:302 %Error: Verilator aborted. Suggest trying --debug --gdbbt
INFO     cocotb:simulator.py:302 %Error: Command Failed ulimit -s unlimited 2>/dev/null; \
    exec verilator_bin -cc --exe -Mdir tb/axis/axis_id/sim_build/test_axis_id \
    -DCOCOTB_SIM=1 \
    --top-module axis_id \
    --vpi \
    --public-flat-rw \
    --prefix Vtop \
    -o axis_id \
    -LDFLAGS -Wl\,-rpath\,/nix/store/bvcp6l3g4ac52gv3fjl8ig1dyfwwhczv-python3-3.9.13-env/lib/python3.9/site-packages/cocotb/libs\ -L/nix/store/bvcp6l3g4ac52gv3fjl8ig1dyfwwhczv-python3-3.9.13-env/lib/python3.9/site-packages/cocotb/libs\ -lcocotbvpi_verilator \   
    -DOBSOLETE \
    -DONESPIN \
    -Wno-fatal \
    --assert -DFORMAL \
    --coverage-line \
    -j 0 \
    -O3 \
    -CFLAGS -O3 \
    --threads 24 \
    -GDATA=16 -GID=0 -GDEST=5 -GUSER=0 -GKEEP=1 -GSTRB=0 -GLAST=1 \
    /nix/store/bvcp6l3g4ac52gv3fjl8ig1dyfwwhczv-python3-3.9.13-env/lib/python3.9/site-packages/cocotb/share/lib/verilator/verilator.cpp \
    .verilator.vlt \
    rtl/axis/axis_id.sv
```

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

The above example is a performed with this cutdown example. Its being exercised with cocotb, using cocotbaxi-ext library. I can provide a tb is required, but everything I'm seeing suggests this isn't related to the actual test

```systemverilog
module axis_id (
    input logic aclk,
    input logic aresetn,
    input logic s_axis_tvalid,
    output logic s_axis_tready,
    input logic s_axis_tdata,
    input logic s_axis_tlast,
    output logic s_axis_tvalid,
    input logic s_axis_tready,
    output logic s_axis_tlast,
    output logic s_axis_tdata
);

always_comb begin
    m_axis_tvalid = s_axis_tvalid;
    m_axis_tdata = s_axis_tdata;
    m_axis_tlast = s_axis_tlast;
    s_axis_tready = m_axis_tready;
end

endmodule
```

ollie-etl, 2024-02-29:
@wsnyder 

> I agree it seems to pop up rarely, perhaps based on CPU allocation etc.

We've a couple of hundred tests which run in ci, but they're cached. If i invalidate the cache to cause a full test suite, then I can guarantee failure, of probably ~5% of our tests. If we didn't cache the results, we'd be unable to use verilator in ci, because we'd never get through a test run

mglb, 2024-03-01:
AFAIK (i.e. in all cases I've seen) the problem is not really relevant for users, because this happens when Verilator is terminating for unrelated reason.

This is caused by use of `exit` when any mutex is locked. Locked mutexes prevent ThreadPool from terminating cleanly, so we've added logic in the destructor that detects which mutexes are locked and unlock them. The error message means that the logic did its work.
It probably would be better to exit without an abort and scary messages, but we don't know what error code has been used to exit from the program.

Anyway, in near future I plan to re-implement all exits from Verilator using an exception. An exception will unwind the stack, run all `V3LockGuard` destructors, and the problem will disappear.

wsnyder, 2024-03-01:
>AFAIK (i.e. in all cases I've seen) the problem is not really relevant for users, because this happens when Verilator is terminating for unrelated reason.

I get this with runs that would normally pass, on normal termination.

If you think it's ok to ignore the message (that is it won't immediately break something later), perhaps make it informational only (UINFO(1)) until we get a rewrite?

ollie-etl, 2024-03-01:
> I get this with runs that would normally pass, on normal termination.

Us too. We'd be fine with this as a warning, but return 0.

closed 2024-03-01 by wsnyder