verilator/verilator #4935: Wrong runtime output on shift reg with OR gate

FurryAcetylCoA, 2024-03-01:
Thank you for making this great project

I have packaged the problematic design into a [dedicated repo](https://github.com/FurryAcetylCoA/verilator-bug1).
In short, the minimized design looks like
``` verilog
module CPU(input wire clock, cpu_reset);
  reg         reset_r;	
  always @(posedge clock) begin	
    reset_r <= cpu_reset;	 // The problematic one
  end 
endmodule

module TOP(input clock, reset);
  reg sync_0, sync_1, sync_2;	
  wire _cpu_reset_chain_io_q = sync_0;	
  CPU cpu ( .clock(clock),
            .cpu_reset(_cpu_reset_chain_io_q | reset) 
  );
  always @(posedge clock) begin
    sync_0 <= sync_1;	
    sync_1 <= sync_2;
    sync_2 <= reset;	
  end
endmodule
```
I tested in `Verilator 5.023 devel rev v5.020-121-ga69cb9b0` and `Verilator 5.022`

This same design also generate an Infinite recursion with flag `-O0`. As described in the repo mentioned.

Please let me know if you need any more info.

Thanks!

FurryAcetylCoA, 2024-03-01:
Thanks for your reply. 

So, from my understanding, the solution to the reset issue is change the test bench from something like
```C++
...
  contextp->timeInc(1);
  dut.release = <Release condition expr>;
  dut.clock = 1;
  dut.eval();
  tfp->dump(contextp->time());
...
```
to
```C++
...
  contextp->timeInc(1);
  dut.release = <Release condition expr>;
  dut.eval(); 
  dut.clock = 1;
  dut.eval();
  tfp->dump(contextp->time());
...
```
Did I get that right?

wsnyder, 2024-03-01:
Change the data bits separate from the posedge clock.  You can do the release than an eval, but that means another eval for the clocks which will be slower.  Assuming you have no negedge clock logic, it's faster to change all data inputs on the negedge.

wsnyder, 2024-03-01:
The issue is really related to your calling your module "TOP". I added a warning for that - please rename.

closed 2024-03-01 by wsnyder