verilator/verilator #4940: Question: 5.018 to 5.022 parameters are treated differently

paul-demo, 2024-03-01:
I'm not sure if this is a bug or not.

I noticed from Verilator version 5.018, I had to access parameters as `dut.MY_PARAM.value.integer` in Cocotb, and in Verilator version 5.022 that is now invalid and I have to use `dut.MY_PARAM.value`.

Is this a VPI interface regression, or a fix, or an unintended consequence of some other change that happened between these two versions?

My version of cocotb has not changed, only Verilator was upgraded.

wsnyder, 2024-03-02:
@marlonjames any comment?

marlonjames, 2024-03-02:
There have been several Verilator VPI changes related to parameters.

Could you provide:
- cocotb version
- the output of `type(dut.MY_PARAM)` for the two versions of Verilator?

paul-demo, 2024-03-04:
cocotb==1.8.1

In Verilator 5.019:
```
type(dut.MY_PARAM) is <class 'cocotb.handle.ConstantObject'>
type(dut.MY_PARAM.value) is <class 'cocotb.binary.BinaryValue'>
```

In Verilator 5.022:
```
type(dut.MY_PARAM) is <class 'cocotb.handle.ConstantObject'>
type(dut.MY_PARAM.value) is <class 'int'>
```

Since it has been changed to an `int`, the `.integer` field no longer exists, so I have a few broken testbenches I need to clean up. I don't particularly care which way it is as long as it stays constant -- maybe the way in 5.022 is preferrable. But, just wanted to let you know that someone changed it!

wsnyder, 2024-03-09:
@marlonjames given above, if you can say this is expected, we can close this issue out.  Thanks.

marlonjames, 2024-03-09:
This is expected behavior on 1.8.1.
The simulation handle code is currently being heavily refactored for cocotb 2.0, so you may want to try using the master branch.

closed 2024-03-09 by wsnyder

paul-demo, 2024-03-11:
I can't use an unstable release. Even what I thought was the stable release (1.8.1) is changing under my feet, as shown above.