verilator/verilator #4941: Address sanitizer fails on verilated_trace_imp.h - load/store to misaligned address

Tikifire, 2024-03-02:
Verilator version
Verilator 5.022 2024-02-24 rev v5.020-157-g2b4852048

OS
WSL2 - Ubuntu22.04 - 5.15.133.1-microsoft-standard-WSL2

I tried to add the argument "--runtime-debug" to "t_trace_cat_fst" from "test_regress" (it would enable the sanitizer), and it gave the results:
```
driver: Leaving directory '/home/tikifire/source-code/verilator/verilator/test_regress/obj_vltmt/t_trace_cat_fst'
        obj_vltmt/t_trace_cat_fst/Vt_trace_cat_fst    > obj_vltmt/t_trace_cat_fst/vlt_sim.log
/home/tikifire/source-code/verilator/verilator/test_regress/../include/gtkwave/fstapi.c:1649:116: runtime error: left shift of negative value -2
/home/tikifire/source-code/verilator/verilator/test_regress/../include/verilated_trace_imp.h:577:69: runtime error: store to misaligned address 0x60d000000114 for type 'uint64_t', which requires 8 byte alignment
0x60d000000114: note: pointer points here
  0c 00 00 00 be be be be  be be be be be be be be  be be be be be be be be  be be be be be be be be
              ^
/home/tikifire/source-code/verilator/verilator/test_regress/../include/verilated_trace_imp.h:196:31: runtime error: load of misaligned address 0x60d000000114 for type 'const uint64_t', which requires 8 byte alignment
0x60d000000114: note: pointer points here
  0c 00 00 00 01 00 00 00  00 00 00 00 0d 00 00 00  c0 02 00 00 70 60 00 00  00 00 00 00 02 00 00 00
              ^
*-* All Finished *-*
```

Actually, I noticed a similar issue - https://github.com/verilator/verilator/issues/3651

gezalore, 2024-03-02:
While I see why this is flagged up. I'm unable to reproduce it with --runtime-debug. What C compiler version are you using?

In general these are harmless on x86

wsnyder, 2024-03-03:
On WSL, using 22.04 with the same compiler you mention, and current master Verilator, running 

        test_regress/t/t_trace_cat_fst.pl --runtime-debug
 
 I don't see this, so please describe how to reproduce.

gezalore, 2024-03-03:
Only happens on `--vltmt`

wsnyder, 2024-03-03:
GTK wave warning reported here gtkwave/gtkwave#313

wsnyder, 2024-03-03:
Verilator's is fixed.  For GTKwave we'll pickup automatically when upstream changes, so just closing this out.

Note as @gezalore indicated these warnings are harmless, didn't affect results.

closed 2024-03-03 by wsnyder