verilator/verilator #4943: Internal Error: .. Canonical variable should be kept or have module refs

ldoolitt, 2024-03-05:
Commit 5e1fc6e2 introduced a new "Internal Error" when processing otherwise well-tested and portable code.
I don't understand what the message is trying to tell me.

```
%Error: Internal Error: oscope_top.sv:142:17: ../V3DfgPasses.cpp:329: Canonical variable should be kept or have module refs
  142 | application_top application_top(
      |                 ^~~~~~~~~~~~~~~
```
This is part of a pretty big code base, and I haven't started trying to reduce it to a usable test case.
If you want to follow along at home, clone https://github.com/BerkeleyLab/Bedrock (master branch is fine).
Then
```
make -C projects/oscope/bmb7_cu Voscope_top_tb
```
Besides Verilator, you'll need yosys and nmigen==0.2,

ldoolitt, 2024-03-05:
Here's the README in this tarball
[issue4943.tar.gz](https://github.com/verilator/verilator/files/14490490/issue4943.tar.gz)

How to reproduce Verilator issue #4943 using files taken from
  https://github.com/BerkeleyLab/Bedrock
A couple of these files are minimally patched.
No weird tools are needed, other than a recent Verilator.

As mentioned in the bug report, Verilator built from commits before 5e1fc6e
works fine:
```
$ sh trigger.sh 
PASS
```
commits 5e1fc6e and later (up to and including today's master 3d5725607) fail:
```
$ sh trigger.sh 
%Error: Internal Error: oscope_top.sv:144:17: ../V3DfgPasses.cpp:324: Canonical variable should be kept or have module refs
  144 | application_top application_top(
      |                 ^~~~~~~~~~~~~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

Sorry this is still a big chunk of code (4488 lines).
Believe me, that's an improvement.  :-/

gezalore, 2024-03-05:
Thanks. I will fix this shortly. -fno-dfg in the meantime if needed.

closed 2024-03-05 by gezalore

gezalore, 2024-03-05:
Now fixed on master. Thanks again for the nice test case.

jersey99, 2024-03-07:
@gezalore Thanks for the quick fix. Out of curiosity, since you must know what triggered the problem. We generate some verilog (via Migen/Litex) and we hand write some of it. Was it just a bug on your side or could our handwritten verilog be better?

gezalore, 2024-03-07:
Hard to say, the reason was a packed array net being partially driven (some bits being driven, some undriven). However, the net in question was an internal signal created when lowering `inout` ports, so I didn't track it all the way back to the input design. You might have a partially driven `inout` port somewhere, maybe called `J18_pmod_a987`, but again I, didn't dig all the way down.

jersey99, 2024-03-15:
Thanks for the pointer @gezalore

commit 3a1355fb (2024-03-05): Fix bad assertion in DFG variable elimination

Fixes #4943