verilator/verilator #4945: Incorrect support for multiple variable definitions in for-loop initializer

nygaard-ibm, 2024-03-06:
It looks like support for multiple variable definitions in a for-loop initializer is incorrect (introduced with d4bb58630efcdfdcfae09bd691561720173a8ffc). I'm trying to use the following syntax (example from SystemVerilog specification 1800-2023 § 12.7.1 The for-loop):
``` verilog
for ( int count = 0, done = 0, j = 0; j * count < 125; j++, count++) $display("Value j = %d\n", j );
```

Verilator appears to accept the following illegal statement: https://github.com/verilator/verilator/blob/0fed5f8b3e02088ee22df4c8b27e6e5e386ce227/test_regress/t/t_for_comma.v#L51

I verified this issue exists in the latest `master` (0fed5f8b3e02088ee22df4c8b27e6e5e386ce227).

```
$ verilator --version
Verilator 5.023 devel rev v5.020-152-g0fed5f8b3

$ sw_vers
ProductName:            macOS
ProductVersion:         14.2.1
BuildVersion:           23C71

$ cat test.sv
module t;
initial for (int a = 0, b = 0; ;) ;
endmodule

$ verilator --lint-only test.sv
%Error: test.sv:2:25: Can't find definition of variable: 'b'
    2 | initial for (int a = 0, b = 0; ;) ;
      |                         ^
%Error: Exiting due to 1 error(s)
```

wsnyder, 2024-03-06:
This test passes on at least 2 of the big 3 simulators.  Why do you think it is illegal?

wsnyder, 2024-03-06:
This test (verilator/test_regress/t/t_for_comma.v) passes on at least 2 of the big 3 simulators. Why do you think it is illegal?

I agree the example you provide should pass - perhaps you'd consider a pull to fix that?

nygaard-ibm, 2024-03-06:
I thought it was illegal because it's not allowed in C. Looking at the spec closer (A.6.8 Looping statements), it looks like it is, my mistake.

If I can carve off some time, I can look into fixing it.

Just wanted to make sure I wasn't doing something wrong or if it was done like this on purpose.

PR #4986 (2024-03-16): Support implicitly-typed variable definitions in for-loop initializers (#4945)
- Adds support for C-style for-loop initializers
    - Current implementation supports: `for (x a = 1, y b = 2, ...)`
    - This patch extends support to:   `for (x a = 1, b = 2, ...)`
    - The implementation is a bit of a hack
- Adds unit test for new feature

This should fix #4945.

Not sure if there is a better way to do this, seems like I'm abusing `yacc`. I couldn't figure out a clean way to clear out `GRAMMARP->m_varDecl` upon entering the `for()`, so I just key off the open parentheses and clear it there.

I [... cut]
closed 2024-03-16 by wsnyder