verilator/verilator #4948: Splitting heavy trace cpp files when --trace-max-array 131072 is used

alexeikom, 2024-03-07:
How may we help - what is your question?
Hello!

I have a small design and I would like to trace 2D arrays. I run Verilator with **--trace-max-array 131072** and this produces two heavy cpp files, 23M each. Each such file has 131072 statements of the form **bufp->chgCData(oldp+25,(vlSymsp->TOP__dut.__PVT__test_data[0]),8);** and **bufp->fullCData(oldp+14661,(vlSymsp->TOP__dut.__PVT__test_data[14658]),8);** 
As a result, gcc takes forever and parallelization is not possible. What can help is splitting those statements to multiple functions and having multiple cpp files upon which make can happily apply parallelization.
Is there an option for packing such statements into function, say, each having 1000 statements? In such a case I'd have a function calling ~132 functions and not 131072. 

I looked into the options and thought that I found the remedy - **--output-split-ctrace**. No, that didn't do the job. **--output-split-cfuncs** didn't help either. I got my best result with **--output-split 1** which split the 2 23MB files into 4 ~11MB each. So that allowed me to involve 2 cores, instead of 1, with make. Nice, but still not enough. The files produced still had this giant number of statements. What can be tried else?

I would be very grateful for any help or guidance!

Thanks,
Alexei

wsnyder, 2024-03-07:
As you discovered there's strong reasons for not setting --trace-max-array that high.  Besides the compile-time issue, potentially every single cycle Verilator is going to do at least 256K loads and 128K compares, which will kill performance.

Instead of doing this, presuming it represents a RAM, can you instead just trace the address and data into the array and debug from that?

If you'd really like array tracing, perhaps you could look at contributing a better algorithm, e.g. one where we look at a min/max address has been changed, and the dump routine then has a non-unrolled loop that looks only within that range to check for changes every cycle?

Another intrum fix is to see if V3Reloop, which rebuilds loops, can optimize this to reduce the amount of code.  Is this something alternatively you might look into?

alexeikom, 2024-03-11:
I will try to see how I can contribute!

alexeikom, 2024-05-15:
Hi!

I might came up with something.

We take for example a "reg [7:0] recv_buf[131071:0]" in "RECV" instance.

My main idea was to add a fake Ast Expression after each place where the memory got update, so

`vlSelf->__PVT__RECV__DOT__recv_buf[__Vdlyvdim0__RECV__DOT__recv_buf__v0] = __Vdlyvval__RECV__DOT__recv_buf__v0;`

will become 

`vlSelf->__PVT__RECV__DOT__recv_buf[__Vdlyvdim0__RECV__DOT__recv_buf__v0] = __Vdlyvval__RECV__DOT__recv_buf__v0;     
 vlSelf->__PVT__RECV__DOT__recv_buf__CHGLST.push_back(__Vdlyvdim0__RECV__DOT__recv_buf__v0);`

This is done via iterating the AstNodes tree just prior to V3EmitC::emitcImp(); call (See code snippet #1 below).

The `__PVT__RECV__DOT__recv_buf__CHGLST` is a vector of unsigned integers (or whatever) that is planted into design header
(in V3EmitCHeaders.cpp in emitAll() call, via another iterating function, see code snippet #2):

`std::vector<uint32_t> __PVT__RECV__DOT__recv_buf__CHGLST;`

In V3EmitCModel.cpp yet another iterating function is used to plant (code snippet #3)

`(vlSymsp->TOP__testxmodem).__PVT__RECV__DOT__recv_buf__CHGLST.clear();`

into the `eval_step()` function to clear the vector of updated indexes at each eval_step() call.

We then replace the huge 
       ` bufp->chgCData(oldp+131232,(vlSymsp->TOP__testxmodem.__PVT__RECV__DOT__recv_buf[0]),8);`
       ` bufp->chgCData(oldp+131233,(vlSymsp->TOP__testxmodem.__PVT__RECV__DOT__recv_buf[1]),8);`
	`.`
	`.`
	`.`
       ` bufp->chgCData(oldp+262303,(vlSymsp->TOP__testxmodem.__PVT__RECV__DOT__recv_buf[131071]),8);`

with 

        for (auto v : vlSymsp->TOP__testxmodem.__PVT__RECV__DOT__recv_buf__CHGLST)
           bufp->chgCData(oldp+v+131232,(vlSymsp->TOP__testxmodem.__PVT__RECV__DOT__recv_buf[v]),8);

This is done with code change in V3EmitCImp.cpp (code snippet #4 below).

I've made a quick test and this seems to do the job. 
Since I'm not very familiar with the Verilator code (looking into it these days), my proposed changes are more of a proof-of-concept, than a usable code update, so excuse me for a dirty code presented here.
I need some guidance about proper places to put the code as well as many other suggestions which I'm sure will follow from you.
I'm attaching the code snippets I used to modify the code. Please review if not too much trouble and we could proceed with this.

Thanks,
Alexei

Code snippets:

**1.**

```
static void iterallassigns(AstNode* nodep, const string &level) {
	int l = 0;
	for (; nodep; nodep = nodep->nextp()) {
		if (VN_IS(nodep, NodeAssign)) {
			if (VN_IS((VN_AS(nodep, NodeAssign))->lhsp(), ArraySel)) {
				AstArraySel *p = VN_CAST((VN_AS(nodep, NodeAssign))->lhsp(), ArraySel);
				// fromp[bitp]
				AstNodeExpr* f = p->fromp();
				AstNodeExpr* b = p->bitp();
				if (VN_IS(f, VarRef) && VN_IS(b, VarRef)) {//AstVarRef is subclass of AstNodeVarRef
					AstVarRef *p = VN_CAST(f, VarRef);
					AstVarRef *p1 = VN_CAST(b, VarRef);
					AstVar* varp = p->varp();
					const AstVar* const varp1 = p1->varp();
[... 234 lines cut ...]
                    } else if (emitTraceIsScBigUint(nodep)) {
                        puts(".get_raw()");
                    } else if (emitTraceIsScBv(nodep)) {
                        puts(")");
                    }
                    puts(")");
                } else {
                    puts("(");
                    iterateConst(nodep->valuep());
                    puts(")");
                }
                ////
                if (emitWidth) puts("," + cvtToStr(nodep->declp()->widthMin()));
                puts(");\n");
        	}
        	else {
				for (int i = 0; i < nodep->declp()->arrayRange().elements(); i++) {
					emitTraceChangeOne(nodep, i);
				}
        	}
        } else {
            emitTraceChangeOne(nodep, -1);
        }
    }
```

wsnyder, 2024-05-18:
Note @gezalore recently committed #5092, which solved a similar-ish problem related to delayed assignments. Basically it creates a queue of updates that are needed.  We should probably do the same thing for tracing - that is for large arrays etc, build a queue of values that tracing needs to check for possible updates.

wsnyder, 2026-10-25:
There's been many trace enhancements and more coming, don't need to leave this open.

closed 2026-10-25 by wsnyder