verilator/verilator #4960: Why do we use TSP sort for variable ordering?

gezalore, 2024-03-10:
Maybe I don't have a good feel for how caches work, but as the title say, I'm curious why we use TSP sort for Variable ordering in multi-threaded mode.

I understand you want to keep variables with the same MTask affinity together to minimize false sharing.

I also understand the TSP sort tries to minimize the "MTask affinity difference" between 2 adjacent blocks. What I don't get is why do we do this? Is this to minimize false sharing on the boundary between two sets of variables with different affinities?

I'm asking because the TSP sort is fairly expensive. It's O(N^2) in space (which is often peak) and slightly worse in time.

wsnyder, 2024-03-10:
Probably because it worked experimentally, @jcoiner?

wsnyder, 2024-05-01:
I think the best we can answer this is the algorithm (by my brief reading only) seems to match that described in our "base" paper *Partitioning and Scheduling Parallel Programs for Multiprocessors*, https://www.cs.rice.edu/~vs3/PDF/Sarkar89.pdf

closed 2024-05-01 by wsnyder