verilator/verilator #4963: The simulation capabilities and application scenarios of Verilator

IceMilkTea, 2024-03-11:
Dear Sir, I would like to ask about the Verilator simulator's support for Verilog syntax: 
1. Why does it not support the deassign syntax and specify syntax in Verilog? 
2.  Regarding the limitation on data state being based on the binary states of '0' and '1 'and not supporting 'x 'and 'z', what is the bottleneck for this restriction? 
Could you please provide some insight?  I am looking forward to your response.
Thank you.

IceMilkTea, 2024-03-12:
I am very grateful for your response! 
From your answer, I understand that Verilator's design is based on practicality and efficiency, which is indeed the case in our usage of Verilator. 
However, there are still some issues, such as when simulating code files from EDA vendor Xilinx. Simulation with Modelsim often goes more smoothly, while Verilator's syntax checking is stricter and requires more modifications to the source files. 
Will Verilator, as an excellent simulation platform, consider optimizing its support in this regard in the future, including support for 'X'\'Z 'states ? 
Looking forward to your reply,thank you.

wsnyder, 2024-03-13:
There will be improvements over time in all areas. I don't personally use Xilinx or FPGAs, nor currently need X/Z, so I am personally unlikely to be spending time on such enhancements.  If they are important to you (or whomever), I'd encourage you (or whomever) to contribute patches towards what you want.

IceMilkTea, 2024-03-13:
I got it.By the way, Verilator is indeed a very promising product, and I look forward to more surprises it will bring.  
Thank you very much.