verilator/verilator #4964: Optimization for Conditional Statement

europe2024, 2024-03-11:
The version of Verilator is Verilator 5.023 devel rev v5.020-171-gba47da658.

The OS is a variant of Debian (version 12).

In the following snippet of code,

     config.vh
     ---------
     localparam UNCORE_RAM_SUPPORTED = 1'b0;

     testbench.sv
     ------------
     `include "config.vh"
          .
          .
          .
      if (UNCORE_RAM_SUPPORTED) sig = testbench.dut.uncore.uncore.ram.ram.memory.RAM[testadrNoBase+i];

if Verilator detects 0 for "expression" in "if (expression) statement;", Verilator should discard "statement" instead of displaying, say, the following error message.

     %Error: testbench.sv:799:84: Can't find definition of 'ram' in dotted signal:
             'testbench.dut.uncore.uncore.ram.ram.memory.RAM'
                                          : ... note: In instance 'testbench'

europe2024, 2024-03-19:
The "if" is not in a generate block because the "if" is not directly within a module.  The "if" is within a task which is within a module. (According to the IEEE Std 1800-2023, "generate regions do not nest, and they may only occur directly within a module.")

Regardless of whether the "if" is in a generate block, when "expression" in "if (expression) statement;" is identically 0, Verilator should discard "statement".

Please consider my comments as a feature request instead of a bug report.

wsnyder, 2024-03-19:
Please post a minimal example that compiles on other simulators.

closed 2024-04-28 by wsnyder