verilator/verilator #4967: Unexpected Significant Performance Degradation on Movement of Code

Karl-Han, 2024-03-12:
This issue is an extension of #4858 . The repository is https://github.com/Karl-Han/cvw/

## Issue Description

There are two versions of testbench code (both of them are equivalent):

- posedge clk version: `if(Validate)` block is put inside an `always @(posedge clk)` block. 
    - on `posedge_clk` branch with `git checkout 73119e3`
- posedge Validate version: `if(Validate)` block is put into another `always @(posedge Validate)` block
    - on `posedge_validate` branch with `git checkout 724483b9`

These two version have with the following difference (the complete code block is attached below):
> git diff posedge_clk posedge_validate

```diff
diff --git a/testbench/testbench.sv b/testbench/testbench.sv
index 7c6cc42ea..61c6ecc4f 100644
--- a/testbench/testbench.sv
+++ b/testbench/testbench.sv
@@ -327,6 +327,9 @@ module testbench;
         $display("Benchmark: coremark is done.");
         $stop;
       end
+  end
+
+  always @(posedge Validate) begin
     if(Validate) begin
       count_validate = count_validate +1;
       if (TEST == "embench") begin
```

In simple words, posedge clk version includes everything inside the `always @(posedge clk)`, while posedge validate version separate the if(Validate) part and put it into `always @(posedge Validate)` block.

The code inside if(Validate) is mostly system calls and file read write.

## Expectation and Runtime Comparison

The expectation was that posedge clk version, while increasing the number of checks performed due to its placement in a clocked process, would not significantly impact overall performance as it is executed the same amount of time as the posedge validate version.

However, the results of posedge validate version have shown a substantial increase in running time, contrary to our expectations. Below are the measured times before and after the modification, in two different modes of operation (more than 40% time difference) :

| mode | posedge clk(s) | posedge Validate(s) |
| --- | --- | --- |
| profiling mode | 1431.83 | 924 |
| non-profile mode | 521.55 | 71.88 |

And the number of validate code block execution is the same for both cases:

posedge clk version in non-profiling mode:
```shell
SUCCESS! All tests ran without failures.
validate =         50
count_clk = testbenchd              445364
- ../testbench/testbench.sv:369: Verilog $finish
501.41user 0.12system 8:21.55elapsed 99%CPU (0avgtext+0avgdata 328960maxresident)k
0inputs+0outputs (0major+80854minor)pagefaults 0swaps
```

posedge validate version in non-profiling mode:
```shell
rv64i_m/I/src/xori-01.S succeeded.  Brilliant!!!
SUCCESS! All tests ran without failures.
validate =         50
count_clk = testbenchd              445363
- ../testbench/testbench.sv:372: Verilog $finish
71.75user 0.13system 1:11.88elapsed 99%CPU (0avgtext+0avgdata 328960maxresident)k
0inputs+0outputs (0major+80823minor)pagefaults 0swaps
```

## Verilator Information

Version: Verilator 5.021 devel rev v5.020-16-g7507c5f56
OS Version: Linux 6.5.0-25-generic 25~22.04.1-Ubuntu
Command: 
```shell
# for non-profiling mode
time verilator -GTEST="\"arch64i\"" -O3 -DVERILATOR=1 --no-trace-top \
	--timescale "1ns/1ns" --timing --binary --top-module testbench \
	"-I../config/shared" "-I../config/rv64gc" ../src/cvw.sv ../testbench/testbench.sv ../testbench/common/*.sv ../src/*/*.sv ../src/*/*/*.sv --relative-includes

# for profiling mode
time verilator -GTEST="\"arch64i\"" --prof-cfuncs -O3 -DVERILATOR=1 --no-trace-top \
	--timescale "1ns/1ns" --timing --binary --top-module testbench \
	"-I../config/shared" "-I../config/rv64gc" ../src/cvw.sv ../testbench/testbench.sv ../testbench/common/*.sv ../src/*/*.sv ../src/*/*/*.sv --relative-includes
```

## Appendix
```systemverilog
  always @(posedge clk) begin
    count_clk = count_clk+1;
    if(SelectTest) begin
      if (riscofTest) memfilename = {pathname, tests[test], "/ref/ref.elf.memfile"};
      else if(TEST == "buildroot") begin 
        memfilename = {RISCV_DIR, "/linux-testvectors/ram.bin"};
        bootmemfilename = {RISCV_DIR, "/linux-testvectors/bootmem.bin"};
      end
      else            memfilename = {pathname, tests[test], ".elf.memfile"};
      if (riscofTest) begin
        ProgramAddrMapFile = {pathname, tests[test], "/ref/ref.elf.objdump.addr"};
        ProgramLabelMapFile = {pathname, tests[test], "/ref/ref.elf.objdump.lab"};
      end else if (TEST == "buildroot") begin
        ProgramAddrMapFile = {RISCV_DIR, "/buildroot/output/images/disassembly/vmlinux.objdump.addr"};
        ProgramLabelMapFile = {RISCV_DIR, "/buildroot/output/images/disassembly/vmlinux.objdump.lab"};
      end else begin
        ProgramAddrMapFile = {pathname, tests[test], ".elf.objdump.addr"};
        ProgramLabelMapFile = {pathname, tests[test], ".elf.objdump.lab"};
      end
      // declare memory labels that interest us, the updateProgramAddrLabelArray task will find 
      // the addr of each label and fill the array. To expand, add more elements to this array 
      // and initialize them to zero (also initilaize them to zero at the start of the next test)
      updateProgramAddrLabelArray(ProgramAddrMapFile, ProgramLabelMapFile, ProgramAddrLabelArray);
    end
    
  ////////////////////////////////////////////////////////////////////////////////
  // Verify the test ran correctly by checking the memory against a known signature.
  ////////////////////////////////////////////////////////////////////////////////
    if(TestBenchReset) test = 1;
    if (TEST == "coremark")
      if (dut.core.priv.priv.EcallFaultM) begin
        $display("Benchmark: coremark is done.");
        $stop;
      end
    if(Validate) begin
      count_validate = count_validate +1;
      if (TEST == "embench") begin
        // Writes contents of begin_signature to .sim.output file
        // this contains instret and cycles for start and end of test run, used by embench 
        // python speed script to calculate embench speed score. 
        // also, begin_signature contains the results of the self checking mechanism, 
        // which will be read by the python script for error checking
        $display("Embench Benchmark: %s is done.", tests[test]);
        if (riscofTest) outputfile = {pathname, tests[test], "/ref/ref.sim.output"};
        else outputfile = {pathname, tests[test], ".sim.output"};
        outputFilePointer = $fopen(outputfile, "w");
        i = 0;
        testadr = ($unsigned(begin_signature_addr))/(P.XLEN/8);
        while ($unsigned(i) < $unsigned(5'd5)) begin
          $fdisplayh(outputFilePointer, DCacheFlushFSM.ShadowRAM[testadr+i]);
          i = i + 1;
        end
        $fclose(outputFilePointer);
        $display("Embench Benchmark: created output file: %s", outputfile);
      end else if (TEST == "coverage64gc") begin
        $display("Coverage tests don't get checked");
      end else begin 
        // for tests with no self checking mechanism, read .signature.output file and compare to check for errors
        // clear signature to prevent contamination from previous tests
        if (!begin_signature_addr)
          $display("begin_signature addr not found in %s", ProgramLabelMapFile);
        else if (TEST != "embench") begin   // *** quick hack for embench.  need a better long term solution
          CheckSignature(pathname, tests[test], riscofTest, begin_signature_addr, errors);
          if(errors > 0) totalerrors = totalerrors + 1;
        end
      end
      test = test + 1; // *** this probably needs to be moved.
      if (test == tests.size()) begin
        if (totalerrors == 0) $display("SUCCESS! All tests ran without failures.");
        else $display("FAIL: %d test programs had errors", totalerrors);
        $display("validate = %d", count_validate);
        $display("count_clk = %ld", count_clk);
`ifdef verilator
        $finish;
`else
        $stop; // if this is changed to $finish, wally-batch.do does not go to the next step to run coverage
`endif
      end
    end
  end
```

davidharrishmc, 2024-03-12:
In other words:

always @(posedge clk)
  if (Validate)
     do something slow

Takes 7 times longer than

always @(posedge Validate)
   do something slow

Even though Validate is only high one cycle at a time so "do something slow" is executed the same number of times in either case.  Printing a message as part of "do something slow" confirms that both blocks invoke the slow task the same number of times.  This is unexpected.

wsnyder, 2024-03-12:
When you mention #4858 is this the same or similar design?  Or just that both discuss performance?

All logic/flops/etc with the same process sensitivity ("clk") can get optimizations such as removal of state, that cannot happen between clocks, as Verilator can't predict the order of separate clocks. So that there's a significant difference is not surprising, though it being as large as 7x is surprising.   Comparing the --stats outputs might give further information.

Karl-Han, 2024-03-12:
> When you mention #4858 is this the same or similar design? Or just that both discuss performance?

They are the same design, and this issue elaborates on the finding of the problem.

- posedge clk version is almost the same as the version mentioned in #4858 except extra variables for observation purpose
- posedge validate version is a modified version of posedge clk version, and the performance gain is out of our expectation.

> All logic/flops/etc with the same process sensitivity ("clk") can get optimizations such as removal of state, that cannot happen between clocks, as Verilator can't predict the order of separate [...]

Command: 
```shell
time verilator -GTEST="\"arch64i\"" -O3 -DVERILATOR=1 --no-trace-top --stats \
	--timescale "1ns/1ns" --timing --binary --top-module testbench \
	"-I../config/shared" "-I../config/rv64gc" ../src/cvw.sv ../testbench/testbench.sv ../testbench/common/*.sv ../src/*/*.sv ../src/*/*/*.sv --relative-includes
```

Here I also attached the compressed stats for you information.
[stats.tar.gz](https://github.com/verilator/verilator/files/14577537/stats.tar.gz)

Below I attached the `diff -bur posedge_clk posedge_validate` between the outputs of two different versions, which includes `Vtestbench__stats_dfg_patterns__post_inline.txt`, `Vtestbench__stats_dfg_patterns__pre_inline.txt` and `Vtestbench__stats.txt`.

Noticeably, there is no abonormal difference between two version. And the total elapsed times are close, i.e., 3.377s for posedge clk version and 3.3635s for validate version. However, there are hugh time difference as above, posedge clk is 17m2s and posedge validate is 2m18s.
```shell
rv64i_m/I/src/xori-01.S succeeded.  Brilliant!!!                     │rv64i_m/I/src/xor-01.S succeeded.  Brilliant!!!
SUCCESS! All tests ran without failures.                             │rv64i_m/I/src/xori-01.S succeeded.  Brilliant!!!
validate =         50                                                │SUCCESS! All tests ran without failures.
count_clk = testbenchd              445364                           │validate =         50
- ../testbench/testbench.sv:369: Verilog $finish                     │count_clk = testbenchd              445363
1020.93user 0.11system 17:02.21elapsed 99%CPU (0avgtext+0avgdata 3281│- ../testbench/testbench.sv:372: Verilog $finish
36maxresident)k                                                      │138.78user 0.10system 2:18.91elapsed 99%CPU (0avgtext+0avgdata 32823
19208inputs+0outputs (0major+80582minor)pagefaults 0swaps            │2maxresident)k
                                                                     │0inputs+0outputs (0major+80590minor)pagefaults 0swaps
```

## stats folder diff

Command: `diff -bur posedge_clk posedge_validate`

```diff
diff --color -bur posedge_clk/Vtestbench__stats.txt posedge_validate/Vtestbench__stats.txt
--- posedge_clk/Vtestbench__stats.txt	2024-03-12 10:43:11.190303389 -0700
+++ posedge_validate/Vtestbench__stats.txt	2024-03-12 11:37:14.850601316 -0700
@@ -193,43 +193,43 @@
   Optimizations, Gate assign merged                                                                   0
   Optimizations, Gate inputs replaced                                                             31539
   Optimizations, Gate sigs deduped                                                                    0
-  Optimizations, Gate sigs deleted                                                                33320
+  Optimizations, Gate sigs deleted                                                                33318
   Optimizations, Inline unsupported                                                                   0
   Optimizations, Inlined instances                                                                 1785
   Optimizations, Lifetime assign deletions                                                         2459
   Optimizations, Lifetime constant prop                                                             100
   Optimizations, Lifetime postassign deletions                                                       23
-  Optimizations, MergeCond longest merge                                                           1918
+  Optimizations, MergeCond longest merge                                                           1963
   Optimizations, MergeCond merged items                                                            5633
-  Optimizations, MergeCond merges                                                                   354
+  Optimizations, MergeCond merges                                                                   353
   Optimizations, Prelim extracted value to ConstPool                                                 31
   Optimizations, Reloop iterations                                                                  128
   Optimizations, Reloops                                                                              2
-  Optimizations, Split always                                                                      4149
+  Optimizations, Split always                                                                      4150
   Optimizations, Substituted temps                                                                13757
   Optimizations, Tables created                                                                      18
   Optimizations, Unrolled Iterations                                                               4609
[... cut ...]
-  Vars, width    32                                                  201        262        271        271
+  Vars, width    32                                                  200        261        270        270
   Vars, width    37                                                    6          8          8          8
   Vars, width    43                                                    1          1          2          2
   Vars, width    44                                                   40         52         52         52
```

wsnyder, 2024-03-13:
Those stat differences seem trivial, which I didn't expect.  More debug would be needed - what does --prof-cfuncs say for both versions? (see Verilator profiling docs for how to use this).

wsnyder, 2024-03-13:
Did/can you use --prof-cfuncs to create this?

Karl-Han, 2024-03-13:
Yes. The commands I use to get `gmon.out`, `gmon.log` and `gmon.log2` are
```shell
$ time verilator -GTEST="\"arch64i\"" --prof-cfuncs -O3 -DVERILATOR=1 --no-trace-top \
	--timescale "1ns/1ns" --timing --binary --top-module testbench \
	"-I../config/shared" "-I../config/rv64gc" ../src/cvw.sv ../testbench/testbench.sv ../testbench/common/*.sv ../src/*/*.sv ../src/*/*/*.sv --relative-includes
$ gprof ./obj_dir/Vtestbench gmon.out > gmon.log
$ verilator_profcfunc gmon.log > gmon.log2
```

Karl-Han, 2024-03-16:
@wsnyder FYI I wrote a proof of concept with a shorter code in https://github.com/Karl-Han/verilator_poc . Hope it can help you to figure out the problem.

gezalore, 2024-04-07:
@Karl-Han Could you please confirm which C++ compiler and version are you using? Running your example https://github.com/Karl-Han/verilator_poc with clang 14.0.0 I see the 2 versions have very similar numbers (33.4s for posedge clk and 32.9s for posedge validate - no profiling). 

With GCC 11.4.0 there is a bigger difference, but posedge clk is actually faster: 48.9s vs 69.7s I suspect we hit something interesting in GCC here.

gezalore, 2024-04-07:
Based on https://github.com/Karl-Han/verilator_poc, the core of the problem is this code 

https://github.com/Karl-Han/verilator_poc/blob/55af19d3191d5cb6dd2ccd20b0641e3d67899d51/posedge_clk/performance_degrade.sv#L51-L54

Slightly rephrased:
```verilog
    int unsigned i = 1;
    while(i != 0) begin
        i++;
        if(i % (2**10) == 1)$fdisplay(fd, "%d\n", i);
    end
```

This compiles into C++ equivalent to this:
https://godbolt.org/z/1coc9efn5

What is going on here, is you loop over all 32-bit integers (!), and do nothing for almost all of them (you only do work for 1 out of 1024), so the time is almost all spent incrementing the loop variable until the 32-bit counter overflows.

As a primary fix, I would recommend not doing that, and enumerate only the values you actually want to compute on if you can (`for (i = 1 ; ... ; i += 1024) ...` or whatever).

I think the reason there is such a big performance difference between the posedge clk and posedge validate version is that by default Verilator compiles output with -Os, and the code generated by GCC with -Os for the above C++ is fairly sensitive to some micro-architectural effect. I'm not entirely sure which one is the specific effect. What I do see is that the emitted x86 loops are identical in the 2 cases, except for register allocation and code alignment, but one of the versions is significantly slower (in my case, posedge validate is slower - measured by `perf record`). In the posedge clk version, all code is in a single C function, while with posedge validate, they are in 2 separate functions, so the precise instruction streams vary, but the loops that take almost all time are identical:

posedge clk:
```asm
3d8:    mov                   ebx, m32@[rbp+0x18]
3db:    test                  ebx, ebx
3dd:    je                    # break loop
3df:    inc                   ebx
3e1:    mov                   eax, ebx
3e3:    mov        m32@[rbp+0x18], ebx
3e6:    and                   eax, 0x3ff
3eb:    dec                   eax
3ed:    jne                   3d8 # loop back
```

posedge validate:
```
47a:    mov                   r8d, m32@[rbx+0x18]
47e:    test                  r8d, r8d
481:    je                    # break loop
483:    inc                   r8d
486:    mov                   eax, r8d
489:    mov        m32@[rbx+0x18], r8d
48d:    and                   eax, 0x3ff
492:    dec                   eax
494:    jne                   47a # loop back
```

There is nothing Verilator can do about this at this point as the code generation is done by the C++ compiler.

You can override the default -Os used by Verilator, either by building in a separate step (not using --binary) and using make OPT_FAST=-O2 a described in the Verilator manual, or by passing `verilator` something like `-CFLAGS -O2`, to override the default. Doing the latter on your example cuts my runtimes from 48.9s/69.7s to a consistent ~17s for both variants.

davidharrishmc, 2024-04-08:
@gezalore I think you are missing the context here.  

The root issue is that our CORE-V Wally testbench is 110x faster when a slow checking step is moved out of the (posedge clk) into (posedge Validate), changing runtime from ~10 minutes to a few seconds.  The example @Karl-Han provided is an attempt to make it easier to reproduce this issue without all the toolchain baggage required to build and run the CORE-V Wally testbench.  As you see, the computation is just a way to waste time to illustrate that the behavior is different with (posedge clk) vs. (posedge something else).   In our actual computation, that computation is replaced by some work reading a bunch of files and comparing expected results. We suspect the root issue might be that the Verilator scheduler improperly does that file reading on every posedge clk even though it should be gated by Validate, then discards the work if Validate is not true.  Moving to (posedge Validate) seems to prevent the work from being done.  Putting print statements in either version print correctly, so the work is taking huge amounts of time in (posedge clk) but not causing incorrect actions.

The compiler isn't the root interesting thing.  We are presently on 13.2.0 and have also used a recent clang to troubleshoot, and different compilers haven't caused more than small percentage changes in speed.

I'm afraid this testbench is too simple and is distracting from the Verilator bug that has a 110x performance difference between the two code formulations.  

We can strip down Wally to a verision that has prebuilt test cases and doesn't need a toolchain. We can express it such that changing one `define will switch between the two (@ posedge) formulations and change the execution time 110x. It is still about 20k lines of SystemVerilog, and I'm afraid this may be overwhelming for debug.

Is there a better strategy to help reproduce and troubleshoot this?

gezalore, 2024-04-08:
If the performance difference is 110x, then running `perf record` on the 2 versions should pretty quickly highlight where the difference lies. It unlikely to be in the generated code, more likely to be the Verilator run-time system (stuff in `include/`). Could you attempt that to see if there is something obvious showing up in the profiles, and maybe propose a fix?

Karl-Han, 2024-04-09:
> If the performance difference is 110x, then running `perf record` on the 2 versions should pretty quickly highlight where the difference lies. It unlikely to be in the generated code, more likely to [...]

As I mentioned in https://github.com/verilator/verilator/issues/4967#issuecomment-1992367061, there are is little difference between two versions. And it turns out that there is no much difference on the profiling reports, even though there is hugh difference in runtime. 

As for the profiling comparison of the PoC, they are in the correspondent folder as `gmon.out`, `gmon.log` and `gmon.log2`. And I don't think there is much difference. 

Therefore, I agree with your idea that it is the problem with run-time system of Verilator. While the same part of code, e.g., if-validate, is called the same amount of time, the runtime difference doesn't make sense at all. It is very likely the event scheduling with NBA is causing the problem.

gezalore, 2024-04-09:
> As for the profiling comparison of the PoC, they are in the correspondent folder as gmon.out, gmon.log and gmon.log2. And I don't think there is much difference.

I explained what is going on with the PoC, if you have tried the suggestion of compiling the generated C++ with -O2 instead of -Os and you have confirmed it does not fix the original simulation, then the PoC is not relevant, so please forget about the PoC.

> As for the profiling comparison of the PoC, they are in the correspondent folder as gmon.out, gmon.log and gmon.log2. And I don't think there is much difference.

Please also forget about --prof-cfuncs and gprof, that flow is supposed to identify user code that is taking most time. The current conjecture is that the problem is not in user code.

Please run your simulation with `perf record <your-simulation-command>`, e.g. `perf record ./obj_dir/Vslowversion` and then run `perf report`. 'perf' is a much more accurate whole program profiler, and if the slowdown is 110x, it will be glaringly obvious where the extra time is spent. It is likely some algorithmic complexity problem. If that is not enough to identify a fix, then we can look at the stripped down 20k line example suggested by @davidharrishmc above.

gezalore, 2024-04-09:
Just to clarify: Please run 'perf' on the exact same simulation that is 110x slower than expected, not the PoC, or anything else.

Karl-Han, 2024-04-15:
Here is the statistics, perf.data files and obj_dir folders for faster and slower versions.

Cloned Repository with `wsim_verilator` branch: https://github.com/Karl-Han/cvw/tree/wsim_verilator

## Summary

| --- | posedge_validate | posedge_clk |
| time | 7.95s | 948.82s |

Attached Files

- intermediate files and executable in `obj_dir`:
    - `obj_dir_fast.tar.gz`: for posedge_validate version
    - `obj_dir_slow.tar.gz`: for posedge_clk version
-  `perf_data.tar.gz`: includes `perf.data` for both versions
    - Result for `perf record ${CVW_HOME/sim/verilator/obj_dir_non_profiling/Vtestbench_rv64gc +TEST=arch64i`

[obj_dir_fast.tar.gz](https://github.com/verilator/verilator/files/14972957/obj_dir_fast.tar.gz)
[obj_dir_slow.tar.gz](https://github.com/verilator/verilator/files/14972958/obj_dir_slow.tar.gz)
[perf_data.tar.gz](https://github.com/verilator/verilator/files/14972959/perf_data.tar.gz)

## Patches

There are two patches for different purposes, and they are attached for your reference.

- `testbench/testbench_dis_uart.diff`: disable uart as uart bug exists
    - it must be patched to `testbench.sv` to get it running on Verilator
- `testbench/testbench_dis_verilator_speed.diff`: difference between `posedge_clk` and `posedge_validate`
    - on `wsim_verilator` branch, it is Faster Version by default

## Commands

```shell
# Compilation
time verilator \
--Mdir obj_dir_non_profiling -o Vtestbench_rv64gc \
-cc --binary \
 -DVERILATOR=1 --no-trace-top --stats \
--timescale "1ns/1ns" --timing --top-module testbench  --relative-includes \
"-I${CVW_HOME}/config/shared" "-I${CVW_HOME}/config/rv64gc" \
wrapper.c \
${CVW_HOME}/src/cvw.sv ${CVW_HOME}/testbench/testbench.sv ${CVW_HOME}/testbench/common/*.sv ${CVW_HOME}/src/*/*.sv ${CVW_HOME}/src/*/*/*.sv

# execution
time ${CVW_HOME}/sim/verilator/obj_dir_non_profiling/Vtestbench_rv64gc +TEST=arch64i
# For posedge_validate version
# SUCCESS! All tests ran without failures.
# - ${CVW_HOME}/testbench/testbench.sv:417: Verilog $finish
#  +TEST=arch64i  7.95s user 0.09s system 99% cpu 8.045 total
# For posedge_clk version
# SUCCESS! All tests ran without failures.
# - ${CVW_HOME}/testbench/testbench.sv:417: Verilog $finish
#  +TEST=arch64i  948.82s user 0.18s system 99% cpu 15:49.25 total

# profiling
perf record ${CVW_HOME}/sim/verilator/obj_dir_non_profiling/Vtestbench_rv64gc +TEST=arch64i
```

## Appendix

### testbench/testbench_dis_uart.diff
```diff
diff --git a/testbench/testbench.sv b/testbench/testbench.sv
index 035e625b..08d58969 100644
--- a/testbench/testbench.sv
+++ b/testbench/testbench.sv
@@ -337,8 +351,8 @@ module testbench;
         memfilename = {RISCV_DIR, "/linux-testvectors/ram.bin"};
         bootmemfilename = {RISCV_DIR, "/linux-testvectors/bootmem.bin"};
         uartoutfilename = {"logs/", TEST, "_uart.out"};
[... 57 lines cut ...]
+//   // in the posedge clk block rather than a separate posedge Validate block.  
+//   // Until it is fixed, provide a silly posedge Validate block to keep Verilator happy.
+//   // https://github.com/verilator/verilator/issues/4967
+//   end // restored
+//   always @(posedge Validate) // added
+// `endif
     if(Validate) begin
       if (TEST == "embench") begin
         // Writes contents of begin_signature to .sim.output file
@@ -420,10 +420,10 @@ module testbench;
 `endif
       end
     end
-`ifndef VERILATOR
-  // Remove this when issue 4967 is resolved and the posedge Validate logic above is removed
+// `ifndef VERILATOR
+//   // Remove this when issue 4967 is resolved and the posedge Validate logic above is removed
   end 
-`endif
+// `endif
 
 
   ////////////////////////////////////////////////////////////////////////////////

```

davidharrishmc, 2024-04-20:
@Karl-Han does the perf_data give you any insight about the differences?

Karl-Han, 2024-04-22:
In the posedge_validate (faster) version, the overheads of NBAs and all other events are distributed evenly, i.e., no specific one of them is consuming a large amount of time.
[image]

On the contrary, in posedge_clk (slower) version, there is a specific nba that consume a great amont of time, which I believe is that always @(posedge clk) block.
[image]

gezalore, 2024-04-23:
I will look into this further. I am assuming I can reproduce the above perf results with the instructions in https://github.com/verilator/verilator/issues/4967#issuecomment-2054277960, but let me know if there is anything else I should be aware of.

davidharrishmc, 2024-04-23:
Unfortunately running tests on CVW involves building the tests from source, which requires a toolchain including riscv64-unknown-elf-gcc, riscof, Sail, etc., which is can be touchy to get everything working.  It would save you a bunch of time if we create a tarball with the compiled arch64i tests.  @Karl-Han can you post a tarball (as small as feasible) and step-by-step directions to reproduce the fast and slow performance results?

gezalore, 2024-04-23:
Thanks, pre-built binaries would indeed help a lot. For analysis, ideally I would need 3 commands per case (here there are 2 cases, the fast one and the slow one): one to run Verilator, one to compile the executable (Verilator output + whatever dpi/c testbench code you might have. Dependencies on libraries available in Ubuntu 22.04 via apt are ok. If you use verilator --binary then you can ignore this.), and one to run the sim with the correct runtime arguments (if any). You don't need to reduce the size of the RTL/SV sources, but eliminating as many dependencies (e.g. target toolchain like you mention) as possible would be great.

gezalore, 2024-04-27:
I can reproduce the slowdown using the above. Note that having to use docker is not tenable. I need my own tools to dig into this, which are not available inside the container. Luckily it works on my native host though, so leave this with me for now.

gezalore, 2024-04-27:
Funny enough, this slowdown is caused by the performance concern I just mentioned in #5072. Here is what's happening.

In the slow case, testbench.sv has code equivalent to this (simplified):

```verilog
task automatic CheckSignature;
   logic [31:0]   sig32[0:5000000];  // <- Large array (20 Mb)
   // ... omitted ...
endtask

always @(posedge clk) begin
  if (Validate) begin  // Rare condition
    Checksignature();
  end
end
```

- V3Task will inline CheckSignature into the always block.
- The scheduling puts the whole always block into a single CFunc.
- V3Localize realizes `sig32` is only used in one CFunc so converts it into a local in that function
- All user signals that are localized are zero initialized on function entry.

The Emitted C++ is something like this:

```cpp
void Vtestbench__nba_sequent_n(state* statep) {
   uint32_t sig32[5000001];
   VL_ZERO(sig32); // this is just memset to 0
   if (statep->Validate) { // Rare condition
      // sig32 is only used here, but always memset above
   }
}
```

That is: we memset 20M of memory on every posedge clock, initializing the localized `sig32` array. (Incidentally the `gcc -Os` code for that memset is also pretty slow, so using -O2 instead helps a lot, but is not the real fix.)

In the fast case, where you have
```verilog
always @(posedge Validate) begin
   CheckSignature();
end
```

This code ends up in a different C++ function:
```cpp
void Vtestbench__nba_sequent_m(state* statep) {
   uint32_t sig32[5000001];
   VL_ZERO(sig32); // this is just memset to 0
   // sig32 is only used here, but always memset above
}
```

However, this C++ function is only called on an actual `@(posedge validate)`, not on every `@(posedge clk)`, so it runs as fast as expected.

@Karl-Han @davidharrishmc as a workaround you can use `verilator -fno-localize`. When #5050 is implemented, you should also be able to change those large arrays to have static lifetime, but that would not help today.

@wsnyder note that this is not just a problem of V3Localize, as in the above, the lifetime of `sig32` is `automatic`, so it indeed needs to be initialized on every entry to the task (it has a defined initial value in SV). We probably should just insert explicit initialization statements for automatic variables where the declaration appears, and not initialize everything on CFunc entry.

wsnyder, 2024-04-27:
>We probably should just insert explicit initialization statements for automatic variables where the declaration appears, and not initialize everything on CFunc entry.

This code would still do badly

```
always @ something begin
   logic [31:0]   sig32[0:5000000];  // <- Large array (20 Mb)
   if (Validate_almost_never_happens) begin  // Rare condition
     do_something_with_sig32;
   end
end
```

We probably should move the zero into the smallest containing block possible.  Currently zero is part of the variable declaration, that would need to be split out.  Alternatively move the variable declaration into arbitrary begin blocks, which is probably easier to read and perhaps better for stack (if different branches have different variable usage).

wsnyder, 2024-04-27:
@Karl-Han @davidharrishmc as another workaround perhaps you can change the offending large array into an associative array?

gezalore, 2024-04-27:
> This code would still do badly
>
> ```
> always @ something begin
>    logic [31:0]   sig32[0:5000000];  // <- Large array (20 Mb)
>    if (Validate_almost_never_happens) begin  // Rare condition [...]

In this code sig32 has static lifetime, so it's not clear, if `do_something_with_sig32` only does partial updates (or reads some bits before it writes those bits), then it cannot be localized in fact. If `do_something_with_sig32` fully writes `sig32` before consuming any bits, then it can be localized and shouldn't need any initialization.

wsnyder, 2024-04-27:
Note V3Life already removes assignments that are not needed, so if clearing was an operation it could optimize it away.  However the hard case is a $readmem which can never be known to set all bits, so I think minimizing the scope for the variable/clear location is useful regardless of clear-lifetime optimization.

davidharrishmc, 2024-04-28:
We don't really need a workaround - the always @(posedge Validate) is a workaround that gives good performance.  We would like to roll back to always @(posedge clk) once the issue is fixed.

wsnyder, 2026-09-13:
#5072 mentioned in this was fixed, and also another workaround was mentioned that gives good performance.  So I'm assuming we're done here.

closed 2026-09-13 by wsnyder