verilator/verilator #4979: Linter allows commas in instance parameters where they are not allowed in Verilog

paul-demo, 2024-03-13:
Verilator's linter is fine with the following trailing comma in parameter list. Most other tools, and I think the language Verilog in general, do not allow a comma after `PARAM_A` here:

```
  some_module #(
      .PARAM_A(1),
  ) some_module_inst (
      .clk(clk),
      .reset(reset)
  );
```

I can probably fix it if you point me to the right files to change.

wsnyder, 2024-03-14:
If you could create a pull to add a test (e.g. t_param_comma_bad.v/.pl/out) that would be appreciated.  Please also check for a similar problem with class parameters.

Empty pins are allowed in the parser and reported later in V3LinkCells, so probably there's a check needed there for parameters.

paul-demo, 2024-03-22:
I didn't see a simple way to fix it. Currently, you look for things with name="" and give them some default param name. However, maybe you also need to check if the value even exists or if it's also empty, and throw a PINNOCONNECT error like you do some of the time.

I didn't get to actually fixing it but wrote down the test cases which succeed even though they should fail:
https://github.com/verilator/verilator/pull/5012

It happens when you have > 1 parameter, and you specify 1 parameter with a comma suggesting you were also filling out the second parameter (though it isn't included). Hopefully that makes sense; see the .v files here:

https://github.com/verilator/verilator/pull/5012

paul-demo, 2024-03-22:
Note: it may be PINNOTFOUND in V3LinkDot.cpp that needs to be changed. That's at least the error message displayed in the other cases.

wsnyder, 2024-04-30:
Thanks for the test cases & looking into this.  I realized we could alternatively change the grammar to fix this, so committed that.

closed 2024-04-30 by wsnyder