verilator/verilator #4981: Is selectable diagnostic message for $finish supported?

dominiksalvet, 2024-03-14:
Is selectable diagnostic message verbosity for `$finish(<val>)`/`$fatal(<val>, ...)` supported? The IEEE 1800-2012 standard shows the following table for the `<val>` argument:

```
0 Prints nothing
1 Prints simulation time and location
2 Prints simulation time, location, and statistics about the mem
```

No matter what number I use, the simulation always prints `- tb/tb.sv:79: Verilog $finish` at the end (using `$finish`).

My Verilator version is `Verilator 4.038 2020-07-11 rev v4.036-114-g0cd4a57ad`

wsnyder, 2024-03-15:
The argument is ignored.  We have some stats I will be adding so wait a week or two, then if you would like this please consider making a pull request.

dominiksalvet, 2024-03-31:
IMHO I would stick with the standard but I do not have broad knowledge of the industry standard simulators, so I understand there is at least pressure from this side (and also it might be practical). Nevertheless, it would be great to have it selectable exactly according to the standard so that users may select the suitable option for them.

Unfortunately, I am busy now, so I cannot prepare a PR. Anyway, thank you for your engagement.

wsnyder, 2024-05-01:
Closing as unlikely to get attention, and limitation is already documented.    If someone contributes a pull it can be reconsidered.

closed 2024-05-01 by wsnyder