verilator/verilator #4982: Short-circuit evaluation ignored for impure functions

esynr3z, 2024-03-15:
Hello,
I've noticed that functions within conditions are always evaluated and short-circuit evaluation does not happen. It is noticeable  by side-effects these functions may introduce.

In the code below `bar` shall not be called:

```verilog
module test;
  function automatic bit foo(bit v);
    $display("foo is called");
    return v;
  endfunction

  function automatic bit bar(bit v);
    $display("bar is called");
    return v;
  endfunction

  initial begin
    if (foo(0) && bar(1)) begin
    end
    $finish();
  end
endmodule
```

The output:

```bash
$ verilator --binary test.sv && ./obj_dir/Vtest
...
foo is called
bar is called
- test.sv:15: Verilog $finish
```

Excerpt from 2017 standard:

>11.3.5 Operator expression short circuiting 
The operators shall follow the associativity rules while evaluating an expression as described in 11.3.2.
Some operators (&&, ||, ->, and ?:) shall use short-circuit evaluation; in other words, some of their
operand expressions shall not be evaluated if their value is not required to determine the final value of the
operation. 

Verilator version: Verilator 5.023 devel rev v5.020-171-gba47da658
OS: Ubuntu 22.04.4 LTS

esynr3z, 2024-03-15:
Oh, I've found already opened #3395 (about ternary operator though) that looks similar. Is it the same issue? Should I close this one?

wsnyder, 2024-03-16:
Yes, #3395 is the same base issue, there's a lot of cleanup needed for short circuting to be completely correct. If you'd like to work on that it would be appreciated.

closed 2024-03-16 by wsnyder