verilator/verilator #4989: error: invalid types'WData {aka unsigned int}[unsigned int]"for array subscript

futurehome, 2024-03-16:
When generating model using verilator, for one of my design, if using "--structs-packed" option, there will be error when compiling the cpp file as shown in below pic; but if without this option, it can be compiled successfully. Any idea what's going on here?

[image]

[image]

What 'verilator' command line do we use to run your example?

What 'verilator --version' are you using?  Did you try it with the git master version?
5.018 
What OS and distribution are you using?
Centos 
May we assist you in trying to fix this in Verilator yourself?

wsnyder, 2024-03-16:
Can you submit a minimal Verilog example that creates this?  As text, please, not screen shot

Note generally --structs-packed was intended for bringing forward designs with old Verilator before unpacked was supported, and it might be that we won't/can't support what is breaking here.

futurehome, 2024-03-18:
The input file:
[error_report.txt](https://github.com/verilator/verilator/files/14633054/error_report.txt)

The command used:
verilator --cc --structs-packed -top-module test error_report.v
make -f obj_dir/Vtest.mk

futurehome, 2024-03-18:
If using command without structs-packed:
verilator --cc --top-module test error_report.v
make -f obj_dir/Vtest.mk

it will report error as below:
[image]

wsnyder, 2024-03-19:
This is because the struct is a primary I/O, if you can avoid that it should work.

futurehome, 2024-03-21:
> This is because the struct is a primary I/O, if you can avoid that it should work.

The design is complicated and using quite a lot struct to communicate between modules.
Are there any difficulties to support it in future release?

wsnyder, 2024-03-21:
Part of it is only because SystemC doesn't support it, for C++ might be straightforward.

Perhaps you'd be willing to look into a pull to support this?

futurehome, 2024-03-21:
> Part of it is only because SystemC doesn't support it, for C++ might be straightforward.
> 
> Perhaps you'd be willing to look into a pull to support this?

Can you point out where should I start? I may have a try.

wsnyder, 2024-03-21:
Here's a brief intro - fully expect you'll be back with more questions, but enough to give something to look at!  Thanks for looking into this.

First, make some test cases that include packed and unpacked structs, include bits, long, quad, >64 bits, strings, array as in your example. You'll need a custom .cpp file with a main like other tests, e.g. t_export_packed_struct.pl?v/cpp.

Then using --debug track down why VL_IN/VL_OUT is being generated instead of the proper data types.  It might be in V3Emit*, or the Ast's might be incorrect - look at the .tree files generated to see if their data types are all correct (e.g. struct remains a struct).  docs/internals.rst describes the format of these files.

futurehome, 2024-04-01:
I'm trying to look at the tree file, but even with the help of internal.rst, I still can't understand what does 1:2:1:4:1:1:1:1:1 mean, can you please explain it? Thanks.

wsnyder, 2024-04-01:
This says what nodes are on the same or different lists.  Specifically indicates which op pointer the node is under from the levels above, e.g. 1:2:1 means this node is directly or a nextp() under some node's op1p, then that parent us under some grandparent's op2p, then some great-grandparent's op1p.