verilator/verilator #4993: NBA order graph of t_timing_nba  is incorrect

gezalore, 2024-03-16:
t_timing_nba has an OrderGraph (attached), containing only NBA Post logic, which does not define any ordering, howerver, there exists an ordering which will make the test fail. This suggests there is some dependency missing. Maybe @kbieganski is interested?
[image]

I will post a follow up on how to reproduce this.

gezalore, 2024-03-16:
Note the OrderGraph is the same before all the refactoring.

gezalore, 2024-03-17:
Assuming #4994 will land, change this line to use `rbegin` instead of `begin`, and t_timing_nba will fail:
https://github.com/gezalore/verilator/blob/98b3e7c835c41902e7aa8725a155ff09fdcef55d/src/V3OrderParallel.cpp#L2444

The only thing that changes is the order we serialize the graph which has no dependenceis.

You can probaly get the same by iterating backwards in serial mode here:
https://github.com/gezalore/verilator/blob/98b3e7c835c41902e7aa8725a155ff09fdcef55d/src/V3OrderSerial.cpp#L48

wsnyder, 2026-09-13:
@gezalore the mentioned line to change to recreate this is long since extinct. Can this issue be closed?

gezalore, 2026-09-15:
On master the graph looks substantially different and no longer fails with arbitrary ordering. Assume fixed along the way. Closing.

closed 2026-09-15 by gezalore