verilator/verilator #4998: Internal error on using constant variable within array assignment

esynr3z, 2024-03-18:
Hello,
I'm getting internal error on the code below. Here I wanted to use constant string within array assignment. It seems legal and works in major simulators.

The code:

```verilog
module test;
  typedef int map_t[string];

  class foo;
    map_t value;
    function new(map_t value);
      this.value = value;
    endfunction
  endclass

  initial begin
    localparam string the_key = "the_answer";
    foo wrp = new('{the_key: 42});
    //foo wrp = new('{"the_answer": 42}); // this works
    $finish();
  end
endmodule
```

The error:

```bash
$ verilator --binary test.sv && ./obj_dir/Vtest
%Error: Internal Error: test.sv:13:19: ../V3Width.cpp:6749: Node has no type
                                     : ... note: In instance 'test'
   13 |     foo wrp = new('{the_key: 42});
      |                   ^~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

Verilator version: Verilator 5.023 devel rev v5.020-195-g6ffff8565
OS: Ubuntu 22.04.4 LTS

wsnyder, 2024-03-18:
Internally this is because the param substitutes TEXT instead of a string constant.

wsnyder, 2024-11-09:
Might be resolved when #5593 is.

wsnyder, 2025-09-13:
Passes on current master, fixed earlier, will push test case to regressions shortly.

closed 2025-09-13 by wsnyder