verilator/verilator #5002: add reflection adjacent capability to model output

ct-clmsn, 2024-03-19:
In order to provide integration with simulation platforms, provide a generic mechanism to get access to I/O port data.

ct-clmsn, 2024-03-19:
proposed solution submitted in PR #5003

wsnyder, 2024-03-19:
I don't understand - why is this needed versus using the VPI?

ct-clmsn, 2024-03-20:
I have a use case where code generated by Verilator is used in a simulator with a plugin system. The verilator code is baked into a plugin. The plugin interface needs to access the verilator generated code's I/O member variables in an opaque manner. Read/write access to verilator's I/O ports is important and this feature allows that access along with providing additional type information that the plugin wrapper needs to make runtime decisions.

wsnyder, 2024-03-20:
And how is what you describe not provided by the VPI?

wsnyder, 2024-03-20:
It's a different mechanism that adds significant maintenance costs which I'm reluctant to add as most users won't need this.  Can you overload more onto the vpi - perhaps something that uses the VPI discovery but has a verilator-specific function to return the raw Verilator data pointer? Then the cost to support is much more acceptable.

wsnyder, 2024-06-15:
Seems discussion ran its course - no change for now.

closed 2024-06-15 by wsnyder