verilator/verilator #5004: "Error-LIFETIME" when using a clocking block inside a submodule

abarajithan11, 2024-03-19:
Example Code:

```
module abc(input clk, input x, output y);

    clocking cb @(posedge clk);
        default input #1ns output #1ns;
        input  x;
        output y;
    endclocking

    task task1;
      @(cb);
      cb.y <= x;
    endtask
    
endmodule

module tb();
    logic clk = 0;
    logic x;
    logic y;

    always #10ns clk = ~clk;

    abc abc(.*);

    initial begin
        repeat(10) abc.task1;
        $finish();
    end
endmodule
```

Command:
```
verilator --binary module_tb.sv
```

Error:
```
%Error-LIFETIME: module_tb.sv:6:16: Invalid reference: Process might outlive variable `y`.                                  : ... note: In instance 'tb.abc'
                                  : ... Suggest use it as read-only to initialize a local copy at the beginning of the process, or declare it as static. It is also possible to refer by reference to objects and their members.
    6 |         output y;
      |                ^
                 ... For error description see https://verilator.org/warn/LIFETIME?v=5.022%Error: Exiting due to 1 error(s)
```

Why I'm doing this:
I have two verification modules [here](https://github.com/abarajithan11/deepsocflow/blob/master/deepsocflow/test/sv/dma.sv) that act as AXI Stream Slave and Master. [Here](https://github.com/abarajithan11/deepsocflow/blob/master/deepsocflow/test/sv/dnn_engine_tb.sv) is how I use them.

wsnyder, 2024-03-19:
@kboronski-ant

wsnyder, 2025-09-09:
This passes on current master, presuming fixed earlier.

closed 2025-09-09 by wsnyder