verilator/verilator #5005: commit 5126689 introduce unpredictable bugs

tastynoob, 2024-03-20:
related issue #4443

i found commit 5126689 cause unpredictable problems in rtl emulation

and this bug exists in subsequent versions

it may produce unpredictable conditional logic and value errors

tastynoob, 2024-03-20:
i think verilator need add large projects to the test set
there must be many more bugs introduced like this

wsnyder, 2024-03-20:
We need a small self-checking test case to know what to do to fix this.

Larger test cases are in https://github.com/verilator/verilator_ext_tests if you have a project that meets the criteria please open a separate issue on that and we'll discuss.

wsnyder, 2024-03-20:
Of course a pull to fix it would also be appreciated (with the small test).

tastynoob, 2024-03-20:
> Of course a pull to fix it would also be appreciated (with the small test).

sure, i will try to fix it

tastynoob, 2024-03-20:
a stupid way to fix it:
add code
`if (node1p->dtypep() != node2p->dtypep()) return false;`
after this
https://github.com/verilator/verilator/blob/88831ca21b97a21dd6a698ec211fa8da331c5c1c/src/V3Ast.cpp#L1101

i found it may produce wrong logic at complex scene
I really have no way to find a simple test to reproduce this bug

in my CPU, this bug will cause a module's output not assigned correctly
i extract this module separately, it can't reproduce
but at least after applying this fix, my CPU works perfectly

tastynoob, 2024-03-20:
NOTE: this fix will cause #4308 test fail
i will find a better way to fix it

tastynoob, 2024-03-20:
it should use:
`if (!VN_IS(node1p->dtypep(), BasicDType) && (node1p->dtypep() != node2p->dtypep())) return false;`
and it works perfectly at my designed CPU and #4308

if you has no objection, i will create one pr

wsnyder, 2024-03-20:
Sure, please make a PR, we'll need a small test with that, thanks!

tastynoob, 2024-03-22:
can i ask about the progress?

wsnyder, 2024-03-22:
Waiting on your pull request.

tastynoob, 2024-03-22:
> Waiting on your pull request.

here #5005

wsnyder, 2024-10-09:
Note this was still open pending a test case, the #5008 pull didn't have one and had other issues.  I'm going to close for now - feel free to reopen if you have a test case - or make a pull with the test case that "passes" with wrong result.

closed 2024-10-09 by wsnyder