verilator/verilator #5009: Unexpected assignment in FSM

scorbetta, 2024-03-22:
Hello all,
I'm having issues while simulating my design using COCOTB and Verilator.

I'm running `Verilator 5.014 2023-08-06 rev v5.014` on a Linux machine mounting `6.5.0-25-generic #25~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Feb 20 16:09:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`.

One portion of the design contains an FSM, whose initial part (the one getting me troubles is reported next)

```
    // States encoding
    localparam READ         = 0;
    localparam WRITE_RESP   = 1;
    localparam WRITE        = 2;
    localparam IDLE         = 3;

    // Internal connections
    reg [$clog2(IDLE)-1:0]  curr_state;

    always @(posedge AXI_ACLK) begin
        if(AXI_ARESETN == 1'b0) begin
            curr_state <= IDLE;
        end
        else begin
            case(curr_state)
```

where it is clear the out-of-reset value of `curr_state`.

However, the simulation gives the following:

file:///home/scorbetta/Pictures/Screenshots/Screenshot%20from%202024-03-22%2012-06-12.png

in which `curr_state` gets the `READ` state instead of `IDLE`. And of course, everything stops working in the design.

The important parameters I pass to Verilator through the COCOTB framework are

```
-Wno-WIDTHEXPAND -Wno-WIDTHTRUNC -Wno-CASEINCOMPLETE --timescale 1ns/100ps --trace-fst --trace-structs
```

What can go wrong?

wsnyder, 2024-03-22:
You have synchronous reset and need to give some clocks while RESETN is low.

closed 2024-03-22 by wsnyder

scorbetta, 2024-03-22:
@wsnyder Ah! Darn it  good catch! Thanks a lot.