verilator/verilator #5011: V3Hash overloaded constructors are ambiguous

paul-demo, 2024-03-22:
MacOS, arm64-apple-darwin22.4.0 doesn't like the V3Hash constructor overloading with compatible types.

```
In file included from ../V3OrderMoveGraph.cpp:23:
../V3OrderMoveGraph.h:57:24: error: call to constructor of 'V3Hash' is ambiguous
                V3Hash hash{reinterpret_cast<uintptr_t>(key.m_domainp)};
                       ^   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/git/verilator/src/obj_dbg/../V3Hash.h:39:14: note: candidate constructor
    explicit V3Hash(uint32_t val)
             ^
/git/verilator/src/obj_dbg/../V3Hash.h:41:14: note: candidate constructor
    explicit V3Hash(int32_t val)
             ^
/git/verilator/src/obj_dbg/../V3Hash.h:43:14: note: candidate constructor
    explicit V3Hash(uint64_t val)
             ^
/git/verilator/src/obj_dbg/../V3Hash.h:45:14: note: candidate constructor
    explicit V3Hash(int64_t val)
             ^
/git/verilator/src/obj_dbg/../V3Hash.h:62:40: error: call to constructor of 'V3Hash' is ambiguous
        return V3Hash{combine(m_value, V3Hash{that}.m_value)};
                                       ^     ~~~~~~
/git/verilator/src/obj_dbg/../V3Hash.h:68:30: note: in instantiation of function template specialization 'V3Hash::operator+<unsigned long>' requested here
        return *this = *this + that;
                             ^
../V3OrderMoveGraph.h:58:22: note: in instantiation of function template specialization 'V3Hash::operator+=<unsigned long>' requested here
                hash += reinterpret_cast<uintptr_t>(key.m_scopep);
                     ^
/git/verilator/src/obj_dbg/../V3Hash.h:39:14: note: candidate constructor
    explicit V3Hash(uint32_t val)
             ^
/git/verilator/src/obj_dbg/../V3Hash.h:41:14: note: candidate constructor
    explicit V3Hash(int32_t val)
             ^
/git/verilator/src/obj_dbg/../V3Hash.h:43:14: note: candidate constructor
    explicit V3Hash(uint64_t val)
             ^
/git/verilator/src/obj_dbg/../V3Hash.h:45:14: note: candidate constructor
    explicit V3Hash(int64_t val)
             ^
2 errors generated.

```

gezalore, 2024-03-22:
Can you please confirm casting the pointer to uint64_t resolves this?

paul-demo, 2024-03-22:
Changing it to uint64_t on those two lines fixes it and allows verilator to compile successfully at top of tree (1c79df86):

[image]

paul-demo, 2024-03-22:
Actually I have another clone which is now failing for this reason too:
https://github.com/verilator/verilator/issues/5013

I think something got updated on my machine and now I'm finding it difficult to build Verilator for all these compilation reasons. Maybe my version of clang/g++ was updated to something newer than what you support?

closed 2024-03-28 by wsnyder