verilator/verilator #5013: out-of-line definition of 'LexerInput' does not match any declaration in 'V3LexerBase'

paul-demo, 2024-03-22:
Apple clang version 14.0.3 (clang-1403.0.22.14.1)
Target: arm64-apple-darwin22.4.0

```
In file included from ../V3ParseLex.cpp:30:
V3Lexer_pregen.yy.cpp:8942:18: error: out-of-line definition of 'LexerInput' does not match any declaration in 'V3LexerBase'
int yyFlexLexer::LexerInput( char* buf, int /* max_size */ )
                 ^~~~~~~~~~
V3Lexer_pregen.yy.cpp:8971:19: error: out-of-line definition of 'LexerOutput' does not match any declaration in 'V3LexerBase'
void yyFlexLexer::LexerOutput( const char* buf, int size )
                  ^~~~~~~~~~~
In file included from ../V3ParseLex.cpp:30:
V3Lexer_pregen.yy.cpp:8942:18: error: out-of-line definition of 'LexerInput' does not match any declaration in 'V3LexerBase'
int yyFlexLexer::LexerInput( char* buf, int /* max_size */ )
                 ^~~~~~~~~~
V3Lexer_pregen.yy.cpp:8971:19: error: out-of-line definition of 'LexerOutput' does not match any declaration in 'V3LexerBase'
void yyFlexLexer::LexerOutput( const char* buf, int size )
                  ^~~~~~~~~~~
2 errors generated.
```

paul-demo, 2024-03-22:
This appears to be a bison/flex version issue, perhaps?
Looks like I have Bison 3.8.2 and Flex 2.6.4

wsnyder, 2024-03-22:
This is unrelated to Verilator, we don't write that code, searching the error gives some hints on you having multiple flex installations and mixing them up, e.g. https://github.com/jonathan-beard/simple_wc_example/issues/7

closed 2024-03-22 by wsnyder

paul-demo, 2024-03-22:
Yeah, I had a package manager update some things against my will; I'll remove that and let's just close this issue.

jackkoenig, 2024-04-20:
For anyone coming across this, you can fix this with `CPPFLAGS="-I/opt/local/include" ./configure` which I learned from an older issue: https://github.com/verilator/verilator/issues/3487#issuecomment-1179412470

Joejiong, 2025-03-06:
> For anyone coming across this using MacOS presumably with MacPorts (or otherwise with flex and bison installed to `/opt/local`), you can fix this with `CPPFLAGS="-I/opt/local/include" ./configure` which I learned from an older issue: [#3487 (comment)](https://github.com/verilator/verilator/issues/3487#issuecomment-1179412470)

you all switched to macPorts?