verilator/verilator #5015: A problem when using DPI-C

yzcccccccccc, 2024-03-23:
When I tried to use DPI-C, I faced this bug report:
> verilator -cc --Mdir ./build/obj_dir --top-module TOP /home/yzcc/ysyx-workbench/npc/build/ErythrinaCore.sv /home/yzcc/ysyx-workbench/npc/build/extern_modules.sv /home/yzcc/ysyx-workbench/npc/build/MemManager.sv /home/yzcc/ysyx-workbench/npc/build/TOP.sv /home/yzcc/ysyx-workbench/npc/build/IFU.sv /home/yzcc/ysyx-workbench/npc/build/MEMU.sv /home/yzcc/ysyx-workbench/npc/build/halter.sv /home/yzcc/ysyx-workbench/npc/build/IDU.sv /home/yzcc/ysyx-workbench/npc/build/EXU.sv /home/yzcc/ysyx-workbench/npc/build/WBU.sv /home/yzcc/ysyx-workbench/npc/build/RegFile.sv /home/yzcc/ysyx-workbench/npc/build/ALU.sv
                    ... Known scopes under 'halt': <no instances found>
%Error: Internal Error: /home/yzcc/ysyx-workbench/npc/build/halter.sv:8:10: ../V3LinkDot.cpp:2731: Couldn't resolve inlined scope 'erythrinacore' in: TOP.erythrinacore.IDU_inst.HaltCtrl
    8 |          halt();
While in `halter.sv` I have:
```
// Generated by CIRCT firtool-1.62.0
module halter(
   input   wire halt_trigger
);
   import "DPI-C" function void halt();
   always @(*) begin
      if (halt_trigger) begin
         halt();
      end
   end
endmodule
```

wsnyder, 2024-03-23:
That code works fine for me, which is not surprising since it doesn't have the code the error mentions. We'll need an example that shows the error and is the minimum possible, you could start by running verilator with --debug and verilating and editing the obj_dir/*__inputs.vpp file, thanks.

yzcccccccccc, 2024-03-25:
The relationship of my system verilog files is: `TOP`->`ErythrinaCore` -> `IDU` -> `halter`. If I set the top module to be `TOP` in verilator, it will trigger the error above. However, if I use `ErythrinaCore` as the top module, verilator can work normally. What's more, if I wrap `TOP` up with another module (let's say `soc`) and set it to be the top module in verilator, it can't work. But in this circumstance, if I set the top module to be `TOP`, verilator can work again. Quite puzzling. BTW, my verilator version is 4.218.

wsnyder, 2024-03-25:
It's hard to say without an example. Perhaps the C code doesn't adjust to the scope name change, or there is a bug in how the scope name is made.

Anyhow we are unlikely to spend any time debugging an old version, try to upgrade.

closed 2024-03-25 by yzcccccccccc