verilator/verilator #5021: Unable to acess verilated.cpp

Abhinandan-Pal, 2024-03-25:
Can you attach an example that shows the issue?  
```shell

root@17a3bf3f6e35:/neuralmc/Again/Verilog# cat toy.sv
module toy (input clk, input rst);

  reg [20:0] c;

  always @(posedge clk)
    if(rst == 1)
      c <= 0;
    else
      c <= c + 1;

endmodule
root@17a3bf3f6e35:/neuralmc/Again/Verilog# verilator --binary -j 0 -Wall toy.sv
make: Entering directory '/neuralmc/Again/Verilog/obj_dir'
make: *** No rule to make target '/home/linuxbrew/.linuxbrew/Cellar/verilator/5.022/share/verilator/include/verilated.cpp', needed by 'verilated.o'.  Stop.
make: Leaving directory '/neuralmc/Again/Verilog/obj_dir'
%Error: make -C obj_dir -f Vtoy.mk -j 8 exited with 2
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /usr/local/bin/verilator_bin --binary -j 0 -Wall toy.sv

``` 
What 'verilator --version' are you using?  Did you try it with the git master version?
```shell
root@17a3bf3f6e35:/neuralmc/Again/Verilog# verilator --version
Verilator 5.022 2024-02-24 rev v5.020-157-g2b4852048
``` 
What OS and distribution are you using?
I am using an Ubuntu docker container where I have built ventilator, the machine I am using is a M2 Macbook Air, running MacOS.

May we assist you in trying to fix this in Verilator yourself?
yes

wsnyder, 2024-03-25:
Verilator wasn't installed correctly. Make sure you are using one of the methods documented, e.g either building and installing without VERILATOR_ROOT, or using it always.  `verilator -V`  will print where the includes are being looked for, which in your case they are not.

Closing but feel free to follow up.

closed 2024-03-25 by wsnyder

Abhinandan-Pal, 2024-03-26:
I followed the "Git Quick Install" on [https://verilator.org/guide/latest/install.html](https://verilator.org/guide/latest/install.html ) and then tried using ventilator right after and also after changing VERILATOR_ROOT following [https://verilator.org/guide/latest/example_binary.html#example-create-binary-execution](https://verilator.org/guide/latest/example_binary.html#example-create-binary-execution) both gave an error indicating verilated.cpp was not found.

verilated.cpp is located in /usr/local/share/verilator/include/ how do I set that as the path to look in?

```shell
root@113f01888bf7:/neuralmc/Again/Verilog# export VERILATOR_ROOT=/usr/local     
root@113f01888bf7:/neuralmc/Again/Verilog# export PATH=$VERILATOR_ROOT/bin:$PATH
root@113f01888bf7:/neuralmc/Again/Verilog# verilator --binary -j 0 toy.sv
make: Entering directory '/neuralmc/Again/Verilog/obj_dir'
make: *** No rule to make target '/home/linuxbrew/.linuxbrew/Cellar/verilator/5.022/share/verilator/include/verilated.cpp', needed by 'verilated.o'.  Stop.
make: Leaving directory '/neuralmc/Again/Verilog/obj_dir'
%Error: make -C obj_dir -f Vtoy.mk -j 8 exited with 2
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /usr/local/bin/verilator_bin --binary -j 0 toy.sv
root@113f01888bf7:/neuralmc/Again/Verilog# verilator -V                  
Verilator 5.022 2024-02-24 rev v5.020-157-g2b4852048

Copyright 2003-2024 by Wilson Snyder.  Verilator is free software; you can
redistribute it and/or modify the Verilator internals under the terms of
either the GNU Lesser General Public License Version 3 or the Perl Artistic
License Version 2.0.

See https://verilator.org for documentation

Summary of configuration:
  Compiled in defaults if not in environment:
    SYSTEMC            = 
    SYSTEMC_ARCH       = 
    SYSTEMC_INCLUDE    = 
    SYSTEMC_LIBDIR     = 
    VERILATOR_ROOT     = /usr/local/share/verilator
    SystemC system-wide = 0

Environment:
    MAKE               = 
    PERL               = 
    SYSTEMC            = 
    SYSTEMC_ARCH       = 
    SYSTEMC_INCLUDE    = 
    SYSTEMC_LIBDIR     = 
    VERILATOR_BIN      = 
    VERILATOR_ROOT     = /usr/local

Supported features (compiled-in or forced by environment):
    COROUTINES         = 1
    SYSTEMC            = 
root@113f01888bf7:/neuralmc/Again/Verilog# cd /usr/local/share/verilator/include/
root@113f01888bf7:/usr/local/share/verilator/include# ls 
gtkwave                verilated_cov_key.h   verilated_intrinsics.h  verilated_sym_props.h  verilated_vcd_c.cpp
verilated.cpp          verilated_dpi.cpp     verilated_probdist.cpp  verilated_syms.h       verilated_vcd_c.h
verilated.h            verilated_dpi.h       verilated_profiler.cpp  verilated_threads.cpp  verilated_vcd_sc.cpp
verilated.mk           verilated_fst_c.cpp   verilated_profiler.h    verilated_threads.h    verilated_vcd_sc.h
verilated.v            verilated_fst_c.h     verilated_save.cpp      verilated_timing.cpp   verilated_vpi.cpp
verilated_config.h     verilated_fst_sc.cpp  verilated_save.h        verilated_timing.h     verilated_vpi.h
verilated_config.h.in  verilated_fst_sc.h    verilated_sc.h          verilated_trace.h      verilatedos.h
verilated_cov.cpp      verilated_funcs.h     verilated_sc_trace.h    verilated_trace_imp.h  vltstd
verilated_cov.h        verilated_imp.h       verilated_std.sv        verilated_types.h
``` 

When I set VERILATOR_ROOT to a path I knew was incorrect the error was different, which I believe indicates in the previous attempt the VERILATOR_ROOT was set correctly in both attempts.

```shell
root@113f01888bf7:/usr/local/share/verilator/include# export VERILATOR_ROOT=/neuralmc/
root@113f01888bf7:/usr/local/share/verilator/include# export PATH=$VERILATOR_ROOT/bin:$PATH
root@113f01888bf7:/usr/local/share/verilator/include# verilator --binary -j 0 toy.sv
sh: 1: exec: /neuralmc//verilator_bin: not found
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /neuralmc//verilator_bin --binary -j 0 toy.sv
```

Abhinandan-Pal, 2024-03-26:
It works now, I had old files inside obj_dir, which was not generated by the verilator installed inside Docker, leading to the incorrect path.
Thank you for the help.