verilator/verilator #5024: Issue when assigning a too wide number in C++

flaviens, 2024-03-27:
Hi there!

First, thank you for your contributions and support!
I believe this is not a new issue, and sorry this may be a duplicate.

For such a design:
```
module top(
    input wire a,
    output wire [2:0] y
);

  assign y[0] = a;
  assign y[2:1] = '0;

endmodule
```

I believe it is quite unintuitive to get the higher-order bits of y non-zero:
```
#include "Vtop.h"

#include <iostream>

typedef Vtop Module;

int main(int argc, char **argv, char **env) {

  Verilated::commandArgs(argc, argv);
  Verilated::traceEverOn(VM_TRACE);

  std::unique_ptr<Module> module_ = std::make_unique<Module>();

  module_->a = 4;
  module_->eval();
  std::cout << "y: " << uint32_t(module_->y) << std::endl;

  exit(0);
}
```

Commands are for example:
```
verilator -Wall -cc top.sv -I.. -I. --exe top.cc -Mdir obj_dir --trace-fst --trace --trace-underscore
make -j 32 -C obj_dir -f Vtop.mk Vtop
./obj_dir/Vtop
```

Thanks!
Flavien

wsnyder, 2024-03-27:
The user's eval code must stay within bounds in what it does.  Use --runtime-debug and it will warn when this happens.

closed 2024-03-27 by wsnyder

flaviens, 2024-03-27:
Sounds reasonable, thank you for your answer!