verilator/verilator #5027: how to trace internal signals of multi insts of a module(hier_block).

panicmarvin, 2024-03-29:
How may we help - what is your question?
I can only find the internal signals the first inst of the hier_block module that has been inst multi times.
verilog code:
```
module detal_code{
    input clk,
    input reset
};
endmodule

module sub_top{
    input clk,
    input reset
};
    detail_code u0{
          .clk(clk),
          .reset(reset)
    };
    detail_code u1{
          .clk(clk),
          .reset(reset)
    };
    detail_code u2{
          .clk(clk),
          .reset(reset)
    };
    detail_code u3{
          .clk(clk),
          .reset(reset)
    };
    detail_code u4{
          .clk(clk),
          .reset(reset)
    };
    detail_code u5{
          .clk(clk),
          .reset(reset)
    };
    detail_code u6{
          .clk(clk),
          .reset(reset)
    };
    detail_code u7{
          .clk(clk),
          .reset(reset)
    };
endmodule

module top{
    input clk,
    input reset
};
    sub_top u0_sub_top{
          .clk(clk),
          .reset(reset)
    };
    sub_top u1_sub_top{
          .clk(clk),
          .reset(reset)
    };
endmodule

```

configure file hier.vlt:
```
`verilator config
hier_block -module "detail_code"
hier_block -module "sub_top"

```
trace code:
```
contextp->traceEverOn(true);
ftp = new VerilatedVcdC;
top->trace(tfp, 99);
tfp->open(“dump.vcd”);
```

cmd
```
verilator --hierarchical --trace  -j -cc -CFLAGS  --top-module top top.sv hier.vlt --exe main.cpp --build
```
I can trace (from VCD) 
```
TOP.top.u0_sub_top.u0.u0 details

but,
missing: TOP.top.u0_sub_top.u1 ~ u7 internal signals
missing: TOP.top.u1_sub_top.u0 ~ u7 internal signals

```
Is it a mistake I made, or does the tool not support this kind of trace?

wsnyder, 2024-03-29:
Your examples have many syntax errors, I cleaned them up but in the future please run the exact files you post through the steps you post, thanks.

The vcd I get after running this is correct.

```
    $scope module u7 $end
     $var wire 1 # clk $end
     $var wire 1 $ reset $end
    $upscope $end
```

I don't know what program is printing the "missing" messages but presumably that tool is the one at fault, or something else.

panicmarvin, 2024-03-30:
Sorry for my mistakes.
I would like to know, for example
```
 $scope module TOP.top.**u1_sub_top** $end
    $scope module u7 $end
     $var wire 1 # clk $end
     $var wire 1 $ reset $end
    $upscope $end
 $upscope $end
```
Can the above-mentioned scope be dumped into a VCD file under the hierarchical configuration I specified?

PR #5029 (2024-03-30): Fix tracing replicated hierarchical models (#5027).
Closes #5027.
closed 2024-03-30 by wsnyder