verilator/verilator #5028: Being able to specify a zero width in an indexed part-select operation violates LRM

chykon, 2024-03-30:
Hello!

IEEE 1800-2017 (11.5.1 Vector bit-select and part-select addressing) requires that the width expression for an indexed part-select operation be a positive integer constant, but Verilator allows the use of 0.

**Example:**

```systemverilog
module tb;
  initial begin
    int val = '1;
    int left;

    $display(val[left+:0]);

    $finish();
  end
endmodule
```

First problem: if the `left` variable is not explicitly initialized, an internal error will occur:

```text
%Error: Internal Error: tb/tb.sv:6:17: ../V3Expand.cpp:398: Width mismatch
    6 |     $display(val[left-:0]);
      |                 ^
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

Second problem: if `left` is initialized, then `$display` will display `x`, although this does not comply with LRM.

**Command:**

```bash
verilator --binary tb/tb.sv
```

---

Verilator version (git master): `Verilator 5.023 devel rev v5.020-223-g39ee522f9`

OS (verilator devcontainer): Ubuntu 22.04.3 LTS

chykon, 2024-03-30:
Another example, but with a different error (in this case, explicit initialization of the variable has no effect):

```systemverilog
module tb;
  initial begin
    int val = '1;
    int left = 0;

    int part = val[left+:0];

    $finish();
  end
endmodule
```

```text
%Error: Internal Error: tb/tb.sv:6:19: ../V3Width.cpp:6446: Under node SEL has no expected width?? Missing Visitor func?
                                     : ... note: In instance 'tb'
    6 |     int part = val[left+:0];
      |                   ^
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

wsnyder, 2024-03-30:
Likely adding an error will fix the second issue.

Perhaps you'd be willing to make a pull request to check for width > 0  (also test negatives)?