verilator/verilator #5030: Fatal error: 'experimental/coroutine' file not found, on macOS using Clang latest

selimsandal, 2024-03-31:
Using LLVM latest from Homebrew or Apple clang version 15.0.0 (clang-1500.3.9.4). I get these errors when I try to create a sim binary.

```
In file included from Vriscv_decoder__pch.h:23:
In file included from ./Vriscv_decoder__Syms.h:17:
In file included from ./Vriscv_decoder___024root.h:9:
/opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.h:39:11: fatal error: 'experimental/coroutine' file not found
# include <experimental/coroutine>
          ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from Vriscv_decoder__pch.h:23:
In file included from ./Vriscv_decoder__Syms.h:17:
In file included from ./Vriscv_decoder___024root.h:9:
/opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.h:39:11: fatal error: 'experimental/coroutine' file not found
# include <experimental/coroutine>
          ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.cpp:23:
/opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.h:39:11: fatal error: 'experimental/coroutine' file not found
# include <experimental/coroutine>
          ^~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
1 error generated.
1 error generated.
```
<br />
I tried changing experimental/coroutine to coroutine. I got these errors.

```
In file included from ./Vriscv_decoder___024root.h:9:
/opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.h:41:23: error: expected namespace name
      using namespace experimental;  // Bring std::experimental into the std namespace
                      ^
In file included from Vriscv_decoder__pch.h:23:
In file included from ./Vriscv_decoder__Syms.h:17:
In file included from ./Vriscv_decoder___024root.h:9:
/opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.h:41:23: error: expected namespace name
      using namespace experimental;  // Bring std::experimental into the std namespace
                      ^
In file included from /opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.cpp:23:
/opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.h:41:23: error: expected namespace name
      using namespace experimental;  // Bring std::experimental into the std namespace
                      ^
/opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.h:100:10: error: no template named 'coroutine_handle' in namespace 'std'
    std::coroutine_handle<> m_coro;  // The wrapped coroutine handle
    ~~~~~^
/opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.h:100:10: error: no template named 'coroutine_handle' in namespace 'std'
    std::coroutine_handle<> m_coro;  // The wrapped coroutine handle
    ~~~~~^
/opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.h:100:10: error: no template named 'coroutine_handle' in namespace 'std'
    std::coroutine_handle<> m_coro;  // The wrapped coroutine handle
    ~~~~~^
/opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.h:114:28: error: no template named 'coroutine_handle' in namespace 'std'
    VlCoroutineHandle(std::coroutine_handle<> coro, VlProcessRef process, VlFileLineDebug fileline)
                      ~~~~~^
/opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.h:114:28: error: no template named 'coroutine_handle' in namespace 'std'
    VlCoroutineHandle(std::coroutine_handle<> coro, VlProcessRef process, VlFileLineDebug fileline)
                      ~~~~~^
/opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.h:114:28: error: no template named 'coroutine_handle' in namespace 'std'
    VlCoroutineHandle(std::coroutine_handle<> coro, VlProcessRef process, VlFileLineDebug fileline)
                      ~~~~~^
/opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.h:201:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) {
                               ~~~~~^
/opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.h:201:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) {
                               ~~~~~^
/opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.h:201:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) {
                               ~~~~~^
/opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.h:269:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) {
[... 123 lines cut ...]
In file included from /opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.cpp:23:
In file included from /opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated_timing.h:28:
In file included from /opt/homebrew/Cellar/verilator/5.022/share/verilator/include/verilated.h:58:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/map:594:
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__tree:2262:26: error: static_cast from '__node_pointer' (aka 'std::__tree_node<std::__value_type<unsigned long long, VlCoroutineHandle>, void *> *') to '__node_base_pointer' (aka 'std::__tree_node_base<void *> *'), which are not related by inheritance, is not allowed
                         static_cast<__node_base_pointer>(__ptr));
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
17 errors generated.
```

<br />
Can you attach an example that shows the issue?
Any 2 modules, one instantiating other.

What 'verilator' command line do we use to run your example?
`verilator --binary -j 0 -Wno-lint --trace '__RTL/riscv_decoder.sv' '__RTL/_TB/riscv_decoder_tb.sv' `

What 'verilator --version' are you using?  Did you try it with the git master version?
Verilator 5.022 2024-02-24 rev UNKNOWN.REV (Homebrew latest), I tried with the master version using (--HEAD)

What OS and distribution are you using?
macOS Sonoma 14.4.1 (23E224)

May we assist you in trying to fix this in Verilator yourself?
Yes

wsnyder, 2024-03-31:
Check the compiler flags are correct, then probably the #ifdefs in verilated_timing.h need to be updated for (yet another) change in what clang requires.  Please send a pull once it's figured out - we'll need to support old versions so add another ifdef on the clang version.

selimsandal, 2024-03-31:
Opened [PR](https://github.com/verilator/verilator/pull/5031) for the fix.

wsnyder, 2024-06-08:
Thanks for everyone's work on this.

closed 2024-06-08 by wsnyder

selimsandal, 2024-06-09:
@wsnyder I installed the latest head (includes the fix [commit](https://github.com/verilator/verilator/commit/9ff06c16649468aec34b71422bc0677e7f3691b0)) and tested the fix, is this issue closed? I still cannot build with coroutines (clang latest)

here are the logs

```
c++ -Os  -I/opt/homebrew/opt/llvm/include -I/opt/homebrew/opt/openjdk/include -I.  -MMD -I/opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include -I/opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-c++11-narrowing -Wno-constant-logical-operand -Wno-non-pod-varargs -Wno-parentheses-equality -Wno-shadow -Wno-sign-compare -Wno-tautological-bitwise-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -DVL_TIME_CONTEXT  -std=gnu++17 -fcoroutines-ts -c -o verilated.o /opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include/verilated.cpp
c++ -Os  -I/opt/homebrew/opt/llvm/include -I/opt/homebrew/opt/openjdk/include -I.  -MMD -I/opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include -I/opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-c++11-narrowing -Wno-constant-logical-operand -Wno-non-pod-varargs -Wno-parentheses-equality -Wno-shadow -Wno-sign-compare -Wno-tautological-bitwise-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -DVL_TIME_CONTEXT  -std=gnu++17 -fcoroutines-ts -c -o verilated_timing.o /opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include/verilated_timing.cpp
In file included from /opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include/verilated_timing.cpp:23:
/opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include/verilated_timing.h:104:10: error: no template named 'coroutine_handle' in namespace 'std'
    std::coroutine_handle<> m_coro;  // The wrapped coroutine handle
    ~~~~~^
/opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include/verilated_timing.h:118:28: error: no template named 'coroutine_handle' in namespace 'std'
    VlCoroutineHandle(std::coroutine_handle<> coro, VlProcessRef process, VlFileLineDebug fileline)
                      ~~~~~^
/opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include/verilated_timing.h:208:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) {
                               ~~~~~^
/opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include/verilated_timing.h:276:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) {
                               ~~~~~^
/opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include/verilated_timing.h:324:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) {
                               ~~~~~^
/opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include/verilated_timing.h:374:29: error: no template named 'coroutine_handle' in namespace 'std'
    void await_suspend(std::coroutine_handle<> coro) const { coro.destroy(); }
                       ~~~~~^
/opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include/verilated_timing.h:409:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) {
                               ~~~~~^
/opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include/verilated_timing.h:426:14: error: no template named 'coroutine_handle' in namespace 'std'
        std::coroutine_handle<> m_continuation;  // Coroutine to resume after this one finishes
        ~~~~~^
/opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include/verilated_timing.h:434:14: error: no type named 'suspend_never' in namespace 'std'
        std::suspend_never initial_suspend() const { return {}; }
        ~~~~~^
/opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include/verilated_timing.h:438:14: error: no type named 'suspend_never' in namespace 'std'
        std::suspend_never final_suspend() noexcept;
        ~~~~~^
/opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include/verilated_timing.h:473:29: error: no template named 'coroutine_handle' in namespace 'std'
    void await_suspend(std::coroutine_handle<> coro) { m_promisep->m_continuation = coro; }
                       ~~~~~^
/opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include/verilated_timing.cpp:237:6: error: no type named 'suspend_never' in namespace 'std'
std::suspend_never VlCoroutine::VlPromise::final_suspend() noexcept {
~~~~~^
In file included from /opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include/verilated_timing.cpp:23:
In file included from /opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/include/verilated_timing.h:28:
[... cut ...]
                         static_cast<__node_base_pointer>(__ptr));
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
16 errors generated.
make: *** [verilated_timing.o] Error 1
%Error: make -C obj_dir -f Vreorder_buffer_tb.mk -j 1 exited with 2
%Error: Command Failed /opt/homebrew/Cellar/verilator/HEAD-962efa3/share/verilator/bin/verilator_bin --binary --timing --top-module reorder_buffer_tb ./rtl/alu32.sv ./rtl/decoder32.sv ./rtl/reorder_buffer.sv ./rtl/tb/reorder_buffer_tb.sv
```

wsnyder, 2024-06-09:
@selimsandal  I don't see any obvious way that the #include isn't happening, but that is what the error suggests, so this seems strange. What does "c++ --version" say?  Are you certain you aren't pointing at old includes or an old verilator?  Otherwise, could you debug this and propose a pull?

selimsandal, 2024-06-09:
c++ version output:
```
Homebrew clang version 18.1.7
Target: arm64-apple-darwin23.5.0
Thread model: posix
InstalledDir: /opt/homebrew/opt/llvm/bin
```
I am using this clang's stdlib; ldflags and cppflags are set correctly.
Verilator version `Verilator 5.025 devel rev v5.024-83-g29db25b70` on macOS
command: `verilator --binary --timing file1.sv file2.sv`

@wsnyder (or anyone that can test this clang + verilator version) can you confirm this sim does not compile?

wsnyder, 2024-06-10:
@selimsandal note I don't use MacOS myself so will need you/others to debug and propose pulls to fix this.

gussmith23, 2024-06-23:
Still occurring on Mac 14.2.1 with default `c++`. (Apple clang version 15.0.0 (clang-1500.3.9.4))

verilator tries to run:
```sh
c++ -Os  -I.  -MMD -I/usr/local/Cellar/verilator/5.026/share/verilator/include -I/usr/local/Cellar/verilator/5.026/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-c++11-narrowing -Wno-constant-logical-operand -Wno-non-pod-varargs -Wno-parentheses-equality -Wno-shadow -Wno-sign-compare -Wno-tautological-bitwise-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -DVL_TIME_CONTEXT  -std=gnu++17 -fcoroutines-ts -c -o verilated_timing.o /usr/local/Cellar/verilator/5.026/share/verilator/include/verilated_timing.cpp
```

If I change -std=gnu++17 to gnu++20, it works.

gussmith23, 2024-06-23:
Happy to help get this fixed up. Will review existing comments, but can you point me to relevant files?

gussmith23, 2024-06-23:
I guess the other option is to keep the std at 17, but enable the specific feature that's needed? I don't know enough about how C++ features are rolled out to know if that maeks sense. I do know that the thing I'm erroring out on is the presence of `coroutine_handle`, which apparently is only available in c++ 20:
https://en.cppreference.com/w/cpp/coroutine/coroutine_handle

gussmith23, 2024-06-23:
Ah, I see it's being worked on in https://github.com/verilator/verilator/issues/5190.