verilator/verilator #5034: Escaped identifiers of VPI variable names have extra space

euphgh, 2024-04-03:
when I use verilator api `contextp->internalsDump();` to print all scopes and variables' name, I found output's escaped identifiers have some difference with "IEEE Standard for SystemVerilog--Unified Hardware Design, Specification, and Verification Language". The minimal exmaple code have shown below, with verilator commit sha 8a4ceb571.

```verilog
module baz (
    input a,
    output b
);
    assign b =! a;
    
endmodule

module foo (
    input a,
    output b
);
    baz inner (
        .a(a),
        .b(b)
    );
endmodule

module \bar (
    input in0,
    output out0
);

    wire \c+b  ;

    foo \first.foo              (
        .a(in0),
        .b(\c+b )
    );

    foo \second.foo (
        .a(\c+b ),
        .b(out0)
    );

    initial begin
        $finish();
    end

endmodule
```
```cpp
#define VM_PREFIX Vour
#define VM_PREFIX_HDR "Vour.h"

#include VM_PREFIX_HDR
#include "verilated.h"
#include "verilated_vpi.h" // Required to get definitions

int main(int argc, char **argv) {
  Verilated::commandArgs(argc, argv);
  const std::unique_ptr<VerilatedContext> contextp{new VerilatedContext};
  const std::unique_ptr<VM_PREFIX> top{new VM_PREFIX{contextp.get(), ""}};

  contextp->internalsDump(); // See scopes to help debug
  while (!contextp->gotFinish()) {
    top->eval();
  }
  return 0;
}
```
```bash
verilator --debug --cc --exe --build --vpi --public-flat-rw our.v main.cc
```

Language specification 5.6.1 section says: **Escaped identifiers shall start with the backslash character (\\) and end with white space (space, tab, newline).** In the following output, however, **verilator append an extra space char for all escaped identifiers.** 

Specifically, with below outputs, I expect module baz's instance's name is `"bar.\first.foo.inner"` instead of `"bar.\first.foo .inner"`, and expect module bar wire variable' name is `"\c+b"` instead of `"\c+b "`. `obj_dir/Vour__Syms.cpp` record all variables and scopes name.

```txt
❯ ./obj_dir/Vour
internalsDump:
  Version: Verilator 5.023 devel
  Argv:
  scopesDump:
    SCOPE 0x558c18314780: TOP
       VAR 0x558c18314d58: in0
       VAR 0x558c18314dd8: out0
    SCOPE 0x558c183147b8: bar
       VAR 0x558c18314e98: \c+b 
       VAR 0x558c18314f18: in0
       VAR 0x558c18314f98: out0
    SCOPE 0x558c183147f0: bar.\first.foo 
       VAR 0x558c18315058: a
       VAR 0x558c183150d8: b
    SCOPE 0x558c18314828: bar.\first.foo .inner
       VAR 0x558c18315198: a
       VAR 0x558c18315218: b
    SCOPE 0x558c18314860: bar.\second.foo 
       VAR 0x558c183152d8: a
       VAR 0x558c18315358: b
    SCOPE 0x558c18314898: bar.\second.foo .inner
       VAR 0x558c18315418: a
       VAR 0x558c18315498: b
```

I would like to kindly inquire whether such behaviors to be in line with fellow developers' expectations?
if yes, I am eager to hear your opinions and suggestions. if no, could you allow me to raise a pull requests (#5033) to fix it ?
Thank you very much for your time and consideration!

wsnyder, 2024-04-03:
I think you are saying that there should be a space and is not, as that is what IEEE requires.

Please ignore what the tree file and internal dump shows, that does not add the extra space and is for internal use; what matters is the VPI-accessed name.  The t_vpi_escape test is intended to check this behavior matches IEEE; a pull that modifies that test to pass on other simulators and fail on Verilator to show the problem, and then fix it would be appreciated.

euphgh, 2024-04-03:
You are right, **what matters is that function `vpi_handle_by_name` can get handle** by escaped identifiers without an extra space. 
But there maybe some misunderstand,  what I am saying is that there should **not** be a space but verilator generates it.

to verify vpi_handle_by_name work, I change main.cc to this
```cpp
#include "vpi_user.h"
#include <iostream>
#define VM_PREFIX Vour
#define VM_PREFIX_HDR "Vour.h"

#include VM_PREFIX_HDR
#include "verilated.h"
#include <string>

void check_name(const char* name) {
  std::string copied_name = name;
  auto hl = vpi_handle_by_name(copied_name.data(), 0);
  if (hl != nullptr) {
    std::cout << "Find \"" << name << "\" by vpi" << std::endl;
  } else {
    std::cout << "Not find \"" << name << "\" by vpi" << std::endl;
  }
}

int main(int argc, char **argv) {
  Verilated::commandArgs(argc, argv);
  const std::unique_ptr<VerilatedContext> contextp{new VerilatedContext};
  const std::unique_ptr<VM_PREFIX> top{new VM_PREFIX{contextp.get(), ""}};

  check_name("bar");
  check_name("bar.\\first.foo");
  check_name("bar.\\first.foo ");
  check_name("bar.\\c+b");
  check_name("bar.\\c+b ");

  while (!contextp->gotFinish()) {
    top->eval();
  }
  return 0;
}
```
result shows escaped identifiers with space can be found but those without space cannot.
personally, I think this behavior is against with IEEE spec. 

```txt
❯ ./obj_dir/Vour
Find "bar" by vpi
Not find "bar.\first.foo" by vpi
Find "bar.\first.foo " by vpi
Not find "bar.\c+b" by vpi
Find "bar.\c+b " by vpi
- our.v:37: Verilog $finish
```

Thank you for your replay

wsnyder, 2024-04-03:
I believe the space is required, please prove otherwise by providing example running on other simulators.

euphgh, 2024-04-08:
OK, now I understand what your means , I will close my issue and pull request

closed 2024-04-08 by euphgh