verilator/verilator #5036: Verilator's vpiStringVal implementation masks byte values of 32 and 0

mjeje, 2024-04-04:
Verilator's vpiStringVal implementation masks byte values of 32 and 0.

verilator version: `Verilator 5.022 2024-02-24 rev UNKNOWN.REV`
OS: macOS Sonoma

Hello. I am writing handler functions to write and read signals from a Verilog Top module. My application uses `vpiStringVal` format. I noticed in the implementation for `vl_get_value(...)` that byte values equal to 0 are returned as spaces (8'b32) to the caller. 

```
//verilated_vpi.cpp:vl_get_value:2349
...
// other simulators replace [leading?] zero chars with spaces, replicate here.
t_outStr[i] = val ? val : ' ';
```

To test an edge case where this mechanic might fail, I wrote a simple Verilog Top module that assigns the values (32 << 8) and 32 to two constant outputs. 32 is represented as the ASCII space character.

```
//Top.v
module Top(output [15:0] const1, output [15:0] const2);
   assign const1 = 16'h20;
   assign const2 = 16'h02;
endmodule;
```

And a test bench that calls the `vl_get_value` wrapper function `vpi_get_value()` for each output. 
The result is two identical char arrays. This behavior is problematic because it is not possible to decode the true value of this output when using `vpiStringVal` format.

```
//TestBench.cpp
...
t_vpi_value const1;
val.format = vpiStringVal;
vpiHandle vh1 = vpi_handle_by_name("const1", NULL);
vpi_get_value(vh1, &const1);
printCharArray(const1.value.str); //{32, 32, 0}
...
t_vpi_value const2;
val.format = vpiStringVal;
vpiHandle vh1 = vpi_handle_by_name("const2", NULL);
vpi_get_value(vh1, &const2);
printCharArray(const2.value.str); //{32, 32, 0}
```

The expected result of printing const1 is `{32, 0, 0}`, and const2 is `{0, 32, 0}`.

wsnyder, 2024-04-04:
Can you please modify one of the Verilator test cases to show this, and also make sure it passes on 2 other simulators and make a pull request or patch with that?  A fix would also be appreciated - should be straight forward once have the tests (to include/verilated_vpi.cpp)!

mjeje, 2024-04-08:
I created tests for these edge cases in test_regress/t/t_vpi_var.cpp. 
https://github.com/mjeje/verilator/tree/vpi-string-val-fix-%235036

These tests fail because vpiStringVal was intended to be used to store ASCII characters for human-readable strings. This explains why leading zeros are replaced by the space character. And why the test macro `TEST_CHECK_CSTR` incorrectly compares a vpi value beginning with `'\0'`, as this marks the end of the string.

While the above functionality is ok for ASCII strings, it limits the ability to use vpiStringVal as a byte array data structure. This format is the most memory space efficient of all the options VPI provides: vpiBinStrVal stores 1 bit per byte, vpiHexStrVal stores 4 bits per byte, and vpiStringVal **stores 8 bits per byte**. For that reason, this use case should be supported.

Adding this change would mean that any code that reads a vpiStringVal could not reliably use the null terminator character to determine the end of the string. From a quick keyword search of the codebase, it looks like this only happens in the tests directory. I can create a PR to add those changes in.

wsnyder, 2024-04-08:
Which other simulators dld you test?  We want it passing on at least 2 others, as this change may break code assuming current behavior, so we need it to be right. Thanks.

mjeje, 2024-04-26:
Would Vivado or Icarus suffice?

wsnyder, 2024-04-26:
Yes, but big-3 preferred.

mjeje, 2024-05-16:
I created a repo to demonstrate this issue in Modelsim, Icarus, and Verilator. All three produced unique outputs when I tested the edge cases above. This may point to ambiguity in the IEEE standard, which allows for an implementation change that resolves these issues.

https://github.com/mjeje/vpiStringVal-bug-demo/tree/main

wsnyder, 2024-05-17:
Thanks for experimenting. Given that, is this then not a bug?

mjeje, 2024-05-17:
That's correct. This is technically not a bug in that it observes IEEE 1800. However, there is evidently some wiggle room in how `vpiStringVal` can be implemented and the current implementation restricts its use to ASCII strings. I am proposing an edit to expand the versatility of this format so that it can be used to store N-width integers. This would allow for memory optimization and more predictable behavior when accessing large hardware logic values.

wsnyder, 2024-05-18:
>I am proposing an edit to expand the versatility of this format so that it can be used to store N-width integers. This would allow for memory optimization and more predictable behavior when accessing large hardware logic values.

This matches at least one of the different other simulators?  If so I'll take the pull request.

mjeje, 2024-05-20:
I do not have access to the other big three simulators, so I can't confirm how VCS and Cadence behave. As for the simulators I tested in the experiment, neither ModelSim nor Icarus allow this usability. However, Verilator's output within the experiment would remain equally distinct (wrt to other simulators) with or without the change.

wsnyder, 2024-05-20:
Ok, please submit the pull to fix this.

wsnyder, 2024-05-24:
Looking at the change proposed, the putVal is now ignoring null characters.  What is passed in is a C string, which must by rules of C be null terminated.  So I don't think we can do this.  Also it looks like most of the simulators that aren't completely broken agree with Verilator.

Questa:
```vpi:     test0=2020
vpi:     test1=ab20
vpi:     test2=20ab
vpi:     test3=20ab
vpi:     test4=2020
```

Xcelium:
```vpi:     test0=0000
vpi:     test1=0000
vpi:     test2=ab00
vpi:     test3=ab00
vpi:     test4=0000
```

Modelsim:
```
vpi:     test0=2020
vpi:     test1=ab20
vpi:     test2=20ab
vpi:     test3=ffab
vpi:     test4=2020
```

wsnyder, 2024-05-24:
I think you'll need to rely on vpiRawTwoStateVal. That should also be portable.  Does that work?

mjeje, 2024-05-24:
I agree. Without providing the length along with the string it's impossible to know which null char indicates the termination of the string. vpi_put_value()'s handling for vpiVectorVal is nearly identical to how vpiStringVal is handled in this change, both are vulnerable to overrun.

Unfortunately, the standard has no room to add a length argument or struct member.

> Looking at the change proposed, the putVal is now ignoring null characters. What is passed in is a C string, which must by rules of C be null terminated. So I don't think we can do this. Also it looks like most of the simulators that aren't completely broken agree with Verilator.
> 
> Questa:
> 
> ```
> vpi:     test1=ab20
> vpi:     test2=20ab
> vpi:     test3=20ab
> vpi:     test4=2020
> ```
> 
> Xcelium:
> 
> ```
> vpi:     test1=0000
> vpi:     test2=ab00
> vpi:     test3=ab00
> vpi:     test4=0000
> ```
> 
> Modelsim:
> 
> ```
> vpi:     test0=2020
> vpi:     test1=ab20
> vpi:     test2=20ab
> vpi:     test3=ffab
> vpi:     test4=2020
> ```

I agree. Without providing the length along with the string it's impossible to know which null char indicates the termination of the string. vpi_put_value()'s handling for vpiVectorVal is nearly identical to how vpiStringVal is handled in this change, both are vulnerable to overrun.

Unfortunately, the standard has no room to add a length argument or struct member.

mjeje, 2024-05-24:
vpiRawTwoStateVal may be a better alternative if it can also be used to read scalar logic values. But the get/put functions for array values are unimplemented in Verilator. Are there plans to add these in the future?

mjeje, 2024-05-24:
> Looking at the change proposed, the putVal is now ignoring null characters. What is passed in is a C string, which must by rules of C be null terminated. So I don't think we can do this. Also it [...]

I agree. Without providing the length along with the string it's impossible to know which null char indicates the termination of the string. vpi_put_value()'s handling for vpiVectorVal is nearly identical to how vpiStringVal is handled in this change, both are vulnerable to overrun.

Unfortunately, the standard has no room to add a length argument or struct member.

wsnyder, 2024-05-24:
>But the get/put functions for array values are unimplemented in Verilator. Are there plans to add these in the future?

Not currently, VPI is basically added as people need a feature, but should be straightforward to make a pull to support them - perhaps something you'd consider?

mjeje, 2024-05-29:
Yes, I can add those functions.

mjeje, 2024-06-17:
I started working on adding the array value accessors. I didn’t realize how much the IEEE standards have changed since 1995 until this week. For instance, the VPI object types `vpiMemory` and `vpiMemoryWord` have been deprecated since 2001, but are still used in the verialtor codebase. As of 1364-2005, Verilog array objects are assigned the type `vpiRegArray` and their elements are `vpiReg`. From what I can tell, the VPI implementation is a mix of different standard definitions.

Of course, replacing `vpiMemory` type with `vpiRegArray` would break backward compatibility because they have different enumerations, 29 and 116 respectively. The standard recommends using compile-time VPI compatibility modes, however, Verilator does not currently have the infrastructure for that.

Below is a list of VPI object types and properties related to the 1800-2023 definitions for `vpi_get_value_array` and `vpi_put_value_array` that are not currently supported in Verilator.
	vpiRegArray (conflicts with vpiMemory)
	vpiReg (conflicts with vpiMemoryWord)
	vpiArrayType
	vpiStaticArray
	vpiArrayNet

I can continue the status quo and use `vpiMemory` and `vpiMemoryWord` in my annex, but this would technically not observe the latest standard. As well as disregard `vpiArrayType`, `vpiArrayNet`, and `vpiStaticArray` by assuming all arrays are static. Alternatively, the addition of compatibility modes would allow for multi-standard support. But this would be a large task.

[image]

wsnyder, 2024-06-17:
The standards are fairly similar since 2005 as you see in the chart, so my inclination would be to make Verilator only support VPI code that adheres to the latest standard.  Any code over 19 years old that violates that table would need to change, but I suspect that's pretty rare and most code will work ok with VpiRegArray versus vpiMemory.

I wouldn't object though to using the version defines as suggested in the standard, but that's more code for us to test/implement, so perhaps we wait until we get a report of older code that breaks?

mjeje, 2024-06-17:
Looking at the incompatibility table, the only descriptions in use by Verilator are related to array handling. Which is within the scope of my PR, so I can append the required replacements.

However, user code that expects vpiMemory or vpiMemoryWord, such as calls to `vpi_get(vpiType,)` will certainly break.

wsnyder, 2024-09-08:
@mjeje, wondering if you plan on finishing off the Raw support?

mjeje, 2024-09-09:
Yes, both raw formats, `vpiIntVal`, and `vpiRealVal`. We plan to open a PR in mid October for the array accessor functions.

wsnyder, 2025-09-06:
vpiRaw* was added in Feb 2025 with #5612, think that covers the final decision on how this should be supported.

closed 2025-09-06 by wsnyder