verilator/verilator #5037: Support generating SAIF file for power estimation

ayanbanrj, 2024-04-05:
Switching Activity Interchange Format (SAIF) files are used by physical implementation tools to perform power estimation and optimization, based on signal switching activity extracted from simulations. It would be great if Verilator could natively generate simulation traces in SAIF format because many tools (for example, Vivado) don't support VCD files as input for power estimation.

The SAIF format is described in IEEE Std 1801-2018, Annex I. Most commercial simulators support SAIF file generation.

> May we assist you in trying to fix this in Verilator yourself?

Yes, I can attempt.

wsnyder, 2024-04-05:
Most power vendors flows have a converter, for example Synopsys' vcd2saif. Do you have that?

While Verilator could do this, is it a better ore general solution to e.g. extend gtkwave to convert formats?  Then all simulators have a open source converter.

wsnyder, 2024-04-28:
Didn't hear back - presuming gtkwave improvement is an option.

closed 2024-04-28 by wsnyder

oharboe, 2025-01-07:
It would be better if verilator supported this directly for large sample runs.

As an example, we're generating a 320gByte .vcd file and the .saif file would probably be a few megabytes as it only contains a summary of the activities. Going via a vcd2saif converter would add many hours extra to such a run and cause problems with artifact sizes with build systems, etc.

OpenSTA can parse .vcd and .saif files.

wsnyder, 2025-01-07:
I worry that this will be infrequently used and be hard to justify the support costs; but if we get enough people saying they will use it, I'll accept a contribution to support this.  Meanwhile, as an alternative can the wave file be dumped by Verilator using `popen` to a pipe command, where that pipe reads the VCD in and writes SAIF out?

BTW another alternative is the toggle coverage report which is likely similar informationally.

oharboe, 2025-01-07:
I see and I recognize that there needs to be a critical mass of users and available resources to implement and support it.

As I understand it and I have found out:

There's no open source vcd2saif command, AFAIK, and even if there was, it would be fantastically inefficient for large runs, say 1000000 samples(a non-trivial workload for a CPU) covering all signals for a small CPU can produce .vcd file that is hundreds of gigabytes compared to megabytes for a .saif file. The .saif file is just a counter per signal, so is efficient to collect and doesn't grow with .vcd size. 

This also explains why there's not a lot of enthusiasm for a vcd2saif converter: it scales terribly.

wsnyder, 2025-01-08:
>There's no open source vcd2saif command

It doesn't matter if there's no open source converter if you aren't using solely open source to consume it.  So are you using an open source tool for power analysis, and what is it?

Anyhow I suggest see if the toggle analysis data has what you want already, in which case a verilated_coverage improvement pull could do conversion, or if not you can start on a pull request to dump SAIF natively.

I'll leave this closed for now as this is only something we will be supporting if we get a contribution.

oharboe, 2025-01-08:
> > There's no open source vcd2saif command
> 
> It doesn't matter if there's no open source converter if you aren't using solely open source to consume it. So are you using an open source tool for power analysis, and what is it?

OpenSTA via OpenROAD-flow-scripts. There is a complete Verilog to GDS-II flow with static and dynamic (read_vcd) power example: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/tree/master/flow/designs/asap7/mock-array

> 
> Anyhow I suggest see if the toggle analysis data has what you want already, in which case a verilated_coverage improvement pull could do conversion, or if not you can start on a pull request to dump SAIF natively.

Will study.

> 
> I'll leave this closed for now as this is only something we will be supporting if we get a contribution.

Yes: until there is a plan and resources, gathering some info in closed status makes sense.

oharboe, 2025-01-09:
I tried `--coverage-toggle` on the [mock-array example](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2666)

Hopefully, I can get some eyes on that to connect the dots between the coverage report and what a .saif file needs to look like.

oharboe, 2025-01-09:
@wsnyder It looks like the necessary information is collected with `--coverage-toggle`...

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2666#issuecomment-2580676792

oharboe, 2025-01-10:
Looking at OpenSTA, there are three things it needs for each pin(bit) from .saif: https://github.com/parallaxsw/OpenSTA/blob/c8e866105046e509d7757d106506e970bf8476ce/power/SaifReader.cc#L182-L185

- The hierarchical name of the pin
- The number of cycles it was logical 1, T1
- The number of toggles, TC

The toggle count is being collected by `--cover-toggle`, but the verilator_coverage does not list the number of cycles the signal was 1.

oharboe, 2025-01-10:
So using `--trace-coverage` for inspiration to implement a new `--trace-saif` option that collects required info and outputs a file?

wsnyder, 2025-01-10:
MoreSaif. look at --trace-vcd, and copy all if the include/*vcd* files to Saif.  Plus add tests.

wsnyder, 2025-01-12:
How large is the FST trace file for your case?  Another possibility is to add a gtkwave converter from FST to SAIF.

I asked my team about exporting SAIF and they generally wouldn't use this and prefer to stay with waves because of the debuggability and implied documentation of saving waves information for the future.  They live with the file size, but suggest maybe FST gets enough of the way there?

Also I assume you are sampling the data - we could add a automatic waves sampling feature (e.g. capture random 1K cycles out of every random 100K)

oharboe, 2025-01-12:
Sold! I think a fst2saif would be a great approach and that it would scale well and has interesting future potential. :-)

> How large is the FST trace file for your case? Another possibility is to add a gtkwave converter from FST to SAIF.

I think a vcd/fst2saif would be great! There's no such open source alternative. Non Verilator users with a vcd2saif need would quickly pick up on this and run vcd2fst foo.vcd | fst2saif foo.saif.

We tried to go this path, but fst2vcd crashed(we need to expand it again to create have OpenSTA process the .vcd file). The .vcd file in our case was 320gByte and the .fst file 13gByte.

As for whether the fst2saif path would be slower than outputting a .saif directly from Verilator, I'm unsure. These days I assume that the limitation is not the number of CPUs, but how effectively one can scale usage. A fst2saif could easily run in a separate process and would probably finish effectively the same time as the .fst file.

I don't have a sense of how much extra simulation time one would get from outputting a 13gByte .fst file compared to outputting a 1mByte .saif file, but I assume that the .fst output is highly optimized whereas the .saif code would not get the same amount of optimization engineering attention.

> I asked my team about exporting SAIF and they generally wouldn't use this and prefer to stay with waves because of the debuggability and implied documentation of saving waves information for the future. They live with the file size, but suggest maybe FST gets enough of the way there?

Got it. I think that we too would benefit from this approach. We'd probably keep these .fst files for a while before letting them expire (deleted automatically) on artifact servers and mark the interesting ones as "keep".

> Also I assume you are sampling the data - we could add a automatic waves sampling feature (e.g. capture random 1K cycles out of every random 100K)

There are many sampling techniques, we are doing a full sample as reference to validate our sampling techniques.

In our case we would prefer to implement our own sampling technique that would build on fst2saif, but we would need a way to merge .saif files, which I think OpenSTA could do directly by taking a list of .saif files instead of a single .saif file in read_saif.

oharboe, 2025-01-12:
Just a datapoint, no surprises really: a Verilator run to generate a 13gbyte .fst file took 8000 seconds and running without .fst generation took 2000 seconds.

wsnyder, 2025-01-12:
There's 4 main slowdowns:
1. Disabled optimizations - needed for any format
2. Time to collect the differences/time deltas per signal - needed for any format
3. Time to make the trace format data - varies by format
4. Time to write the data - varies by format, ultimately limited by storage performance

Use --threads to parallelize most of these; currently 3 and 4 don't parallelize but see https://github.com/gtkwave/gtkwave/issues/286

As to the conversion, I would suggest contribute a new fst2saif to GTKwave.  It needs GTKwave anyways for the fst reading, and it already has fst2vcd etc as a template.

oharboe, 2025-01-12:
Possibly https://github.com/gtkwave/gtkwave/blob/master/src/helpers/fstminer.c would be a better starting point.

wsnyder, 2025-01-12:
Perhaps, perhaps time to open a GTKwave issue and discuss with them.

oharboe, 2025-01-13:
No suggestions, just some data before trying --threads:

simulation without vcd/fst generation 2000 seconds
generate vcd 8000 seconds
generate fst 11000 seconds
fst2vcd 21222 seconds

Lots of room for improvement, but while we're waiting for those, we'll generate a giant .vcd, but we won't store it as an artifact. OpenSTA memory usage is currently, but not for long, a function of .vcd size.

oharboe, 2025-01-13:
> Are you using --threads? FST compression is likely slowing FST.

Not yet, just making a note of "before" numbers.

oharboe, 2025-01-25:
> Are you using --threads? FST compression is likely slowing FST.

We're getting a segfault, which I'm pretty certain is entirely in our code, when we use `--threads`, so we're letting that sleeping dog lie for now.