verilator/verilator #5039: make: *** No rule to make target '/usr/local/share/verilator/include/verilated.cpp', needed by 'verilated.o'.  Stop.

abarajithan11, 2024-04-06:
OS & distribution : WSL2, Ubuntu 22.04, 5.15.146.1-microsoft-standard-WSL2
Verilator version: Verilator 5.024 2024-04-05 rev v5.024-42-gc561fe8ba
GCC Verison: gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0

I had some issues with installing and getting verilator running. I finally decided to go with running from git repo.

What I did:
```
git clone https://github.com/verilator/verilator.git
cd verilator
make
export VERILATOR_ROOT=/home/aba/verilator
```

The command I use:
```
$VERILATOR_ROOT/bin/verilator --binary -j 0 -O3 --trace --relative-includes --top dnn_engine_tb -I../ -F ../sources.txt -CFLAGS -I../ /mnt/d/dnn-engine/deepsocflow/c/sim.c --Mdir ./
```
The error I get:
```
make: Entering directory '/mnt/d/dnn-engine/run/work/build'
make: *** No rule to make target '/usr/local/share/verilator/include/verilated.cpp', needed by 'verilated.o'.  Stop.
make: Leaving directory '/mnt/d/dnn-engine/run/work/build'
%Error: make -C ./ -f Vdnn_engine_tb.mk -j 20 exited with 2
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /home/aba/verilator/bin/verilator_bin --binary -j 0 -O3 --trace --relative-includes --top dnn_engine_tb -I../ -F ../sources.txt -CFLAGS -I../ /mnt/d/dnn-engine/deepsocflow/c/sim.c --Mdir ./
```

Is this an incompatibility with GCC version?

wsnyder, 2024-04-06:
What does `$VERILATOR_ROOT/bin/verilator -V` print?  

I suspect you have an old build directory with a .d file pointing at that old include.  Delete all of your tests' build directories and try again.

abarajithan11, 2024-04-06:
```
Verilator 5.024 2024-04-05 rev v5.024-42-gc561fe8ba

Copyright 2003-2024 by Wilson Snyder.  Verilator is free software; you can
redistribute it and/or modify the Verilator internals under the terms of
either the GNU Lesser General Public License Version 3 or the Perl Artistic
License Version 2.0.

See https://verilator.org for documentation

Summary of configuration:
  Compiled in defaults if not in environment:
    SYSTEMC            =
    SYSTEMC_ARCH       =
    SYSTEMC_INCLUDE    =
    SYSTEMC_LIBDIR     =
    VERILATOR_ROOT     = /opt/verilator/share/verilator
    SystemC system-wide = 0

Environment:
    MAKE               =
    PERL               =
    SYSTEMC            =
    SYSTEMC_ARCH       =
    SYSTEMC_INCLUDE    =
    SYSTEMC_LIBDIR     =
    VERILATOR_BIN      =
    VERILATOR_ROOT     = /home/aba/verilator

Supported features (compiled-in or forced by environment):
    COROUTINES         = 1
    SYSTEMC            =
```

wsnyder, 2024-04-06:
`    VERILATOR_ROOT     = /opt/verilator/share/verilator`

So if you have a reference to /usr/local/share/verilator/include/verilated.cpp you must have an old build file around, or are running the wrong Verilator.

closed 2024-04-06 by wsnyder

abarajithan11, 2024-04-06:
Thank you, deleting the build directory fixed the issue. 
I wonder how I can get rid of old verilator completely and install the latest in a way I can simply call "verilator"
I've done apt-get remove, but it hasn't deleted the files in opt/verilator

wsnyder, 2024-04-06:
Build it without VERILATOR_ROOT set and witha --prefix, see the installation instructions.