verilator/verilator #5041: Verilator not handling hierarchical signal references correctly

slmnemo, 2024-04-09:
Thanks for taking the time to report this.

This git issue details the problem. https://github.com/openhwgroup/cvw/issues/715

Once building our repo's tests and sourcing cvw/setup.sh, running `cd ${WALLY}/sim/verilator && verilator -GTEST="\"arch64i\"" -DVERILATOR=1 --timescale "1ns/1ns" --timing --binary --top-module testbench -I${WALLY}/config/shared -I${WALLY}/config/rv64gc ${WALLY}/src/cvw.sv ${WALLY}/testbench/testbench.sv ${WALLY}/testbench/common/*.sv ${WALLY}/src/*/*.sv ${WALLY}/src/*/*/*.sv --relative-includes` should cause an error related to not finding an instance which is known to exist in other EDA tools.

The exact path is dut.uncore.uncore.uart.uart.u.[signalname]. Expected versus actual behavior is as follows

- Use `dut.uncore.uncore.uart.uart.uart.uart.uart.uart.PCLK`. Expected: Errors when it runs out of "uart" instance names and points to bad location. Actual: Does not complain at all.
- Try to detect `dut.uncore.uncore.uart.uart.UARTSin` Expected: Does not detect it as we are inside the uart_apb.sv module which maps UARTSin to SIN, causing an error Actual: Works just fine
- Rename uart_apb to use uart2 and then use `dut.uncore.uncore.uart.uart2.PCLK` Expected: Finds PCLK. Actual: Fails to find uart2 and then fails to find PCLK.

I suspect the nested if generate blocks are causing issues, as simply trying to access `dut.uncore.uncore.uart.uart` finds a BEGIN rather than a CELL, while `dut.uncore.uncore` results in the statement finding a CELL.

Replication steps:
> Make three-layer hierarchical model, DUT -> if generate begin : A -> CELL B -> if generate begin : C -> CELL D.
> Try to reference CELL D in a testbench for the DUT using DUT.A.B.C.D.
> DUT.A.B.C.D results in error where dotted variable D does not exist
> DUT.A.B.C.C... results in error where C is a BEGIN statement but Verilator expects a variable.

Verilator 5.022 2024-02-24 rev v5.020-157-g2b4852048

Ubuntu 22.04

Yes, you may help me fix this if it's an issue on our end.

davidharrishmc, 2024-04-09:
Kaitlin, I think there is more to this than you yet know.  Rose and Kunlin have been working on hierarchical references and there are some subtle issues.  For example, the Verilog specification prohibits code containing a hierarchical reference to a conditionally generated block that is not instantiated, even if the reference is also conditionally generated and not live in this configuration.  VCS and Verilator seem to adhere to this spec, while Questa is more liberal and allows the reference.  

Let's put this issue on hold until we understand it better.  Please check in with Rose.

wsnyder, 2025-09-09:
Assuming this got resolved, given the last comment about being on hold, otherwise please open a new issue or reopen, thanks.

closed 2025-09-09 by wsnyder

davidharrishmc, 2025-09-09:
Thank you.  We have a workaround.