verilator/verilator #5047: Spurious "VERILATOR_ROOT is set to inconsistent path" errors

polmacanceart, 2024-04-12:
I have downloaded and compiled the latest version of verilator. I am naming this "5.24-dev". I install it to:
`/proj/ipn_verification/wright/ipn_verification/work/software/verilator/5.24-dev`

I set VERILATOR_ROOT accordingly:

```
> % which verilator
/proj/ipn_verification/wright/ipn_verification/work/software/verilator/5.24-dev/bin/verilator
> % echo $VERILATOR_ROOT
/proj/ipn_verification/wright/ipn_verification/work/software/verilator/5.24-dev

```
Now run it:
```
verilator -version
%Error: verilator: VERILATOR_ROOT is set to inconsistent path. Suggest leaving it unset.
%Error: VERILATOR_ROOT=/proj/ipn_verification/wright/ipn_verification/work/software/verilator/5.24-dev

```
Looking at the announcements, I see that something changed: https://github.com/verilator/verilator/pull/4927

I see that bin/verilator now redirects to share/verilator/bin/verilator.

So I edited that file to get it to tell me what it thinks VERILATOR_ROOT should be:

```
%Error: verilator: VERILATOR_ROOT is set to inconsistent path. Suggest leaving it unset.
%Error: verilator: verilator_root is /proj/ipn_verification/wright/ipn_verification/work/software/verilator/5.24-dev/share/verilator
%Error: VERILATOR_ROOT=/proj/ipn_verification/wright/ipn_verification/work/software/verilator/5.24-dev

```

So this file thinks it should still be in the top-level bin directory and not under share?

Shouldn't this script now understand that it's under share/verilator/bin and not bin?

gezalore, 2024-04-12:
VERILATOR_ROOT is intended to point to the directory that contains the `bin` and `include` subdirectories, like in a source tree. This was always the intention, even before the change, and is describe in the installation manual here: https://github.com/verilator/verilator/blob/master/docs/guide/install.rst#1-run-in-place-from-verilator_root

You are using installation method 2 from here: https://github.com/verilator/verilator/blob/master/docs/guide/install.rst#2-install-into-a-specific-prefix Albeit it's not explicitly said, this does not require setting VERILATOR_ROOT. The scripts know what it should be, and hence the error if you set it to something inconsistent. As the method suggests, you should leave VERILATOR_ROOT unset and just run the `verilator` executable, it should take care of everything else for you.

If you want to know what VERILATOR_ROOT is (though ideally you shouldn't have to care), run `../install/bin/verilator --getenv VERILATOR_ROOT`

Closing as answered, but please follow up if needed.

closed 2024-04-12 by gezalore

wa-kakalala, 2024-09-28:
I have the same problem.
```
verilator -version 
[Output]
%Error: verilator: VERILATOR_ROOT is set to inconsistent path. Suggest leaving it unset.
%Error: VERILATOR_ROOT=/home/wkk/software/oss-cad-suite/share/verilator
```
and，i just use `unset VERILATOR_ROOT`, it can work correctly.
```
verilator -version
[Output]
Verilator 5.029 devel rev v5.028-94-g371a4055b
```

wsnyder, 2024-09-28:
@wa-kakalala, not clear what your question is? But it looks like the message is doing as intended, ou should be unsetting to prevent problems.