verilator/verilator #5048: Support out of order package compilation

AndrewNolte, 2024-04-13:
I understand that this is a part of the LRM as described by PKGNODECL, but this would be very useful for lint tools, perhaps only in --lint-only mode in combination with -y. Right now if this happens it sometimes shows up as `syntax error, unexpected IDENTIFIER, expecting TYPE-IDENTIFIER` with something like Pkg::my_type_t

wsnyder, 2024-04-13:
I appreciate the usefulness. Packages being declared, along with any data types imported from packages, is fundamental to the current parser.  Fixing this is likely a month of work.  If you or someone would like to do it we'll certainly take a pull request, but not something that will otherwise happen in any reasonable time, so closing.

closed 2024-04-13 by wsnyder

AndrewNolte, 2024-04-15:
Ok, I think I can probably do a workaround with rerunning the tool with the missing packages prefixed based on the output. There's still the issue where it ends up being "syntax error" instead of PKGNODECL. I'll post another issue if I can get a small reproducer of that

AndrewNolte, 2024-04-15:
This is for use in this extension https://github.com/AndrewNolte/vscode-system-verilog