verilator/verilator #5049: Array locator methods `min` and `max` return wrong values on a fixed-size unpacked array

goekce, 2024-04-15:
#### Example

```sv
module m;
  int unsigned x[3] = '{0, 1, 2};
  initial begin
    $display(x.min);
    $finish;
  end
endmodule
```

Run:

```sh
verilator -MAKEFLAGS '-s' --quiet --binary m.sv && obj_dir/Vm +verilator+quiet
```

Result:

```
2302531392            (comment: changes after every execution)
- m.sv:5: Verilog $finish
```

Expected:

```
'{'h0}        (comment: array locator methods return a queue)
- m.sv:5: Verilog $finish
```

Version and OS

```
Verilator 5.024 2024-04-05
Archlinux
```
---

#### Notes

I could only find tests for fixed-sized size unpacked arrays but:
1. for associative arrays

     https://github.com/verilator/verilator/blob/1315aa31ed1b980b05778f5a0a1f3ff9718be680/test_regress/t/t_assoc_method.v#L90

2. for queues
   
      https://github.com/verilator/verilator/blob/1315aa31ed1b980b05778f5a0a1f3ff9718be680/test_regress/t/t_queue_method.v#L161

So using a queue instead works as expected:

```sv
module m;
  int unsigned x[$] = '{0, 1, 2};
  initial begin
    $display(x.min);
    $finish;
  end
endmodule
```
Result:

```
'{'h0}
- m.sv:5: Verilog $finish
```

`min` and `max` are defined in *LRM 2023 7.12.1 Array locator methods*. AFAI understand these must be defined for all arrays.

---

Currently I don't have the capacity for a PR other than preparing a test case.

wsnyder, 2024-04-15:
min and max return queues per IEEE, so Verilator is correct.  Some other simulators either just print a number like Verilator does, and another gives an error which is to "use display with a %p format".   I'll change this issue to be to add a similar error message.

goekce, 2024-04-16:
I am confused :/

> min and max return queues per IEEE

But why does `{unpacked static array}.min` does not return a queue then? My second (queue) example is displayed correctly without a need for `%p` specifier. IMHO the problem  is the return type of my first example (unpacked static array). If I print the typename:

```sv
module m;
  int unsigned x[3] = {0, 1, 2};
  initial begin
    $display($typename(x.min));
    $finish;
  end
endmodule
```
```sh
$ verilator -MAKEFLAGS '-s' --quiet --binary m.sv && obj_dir/Vm +verilator+quiet
...
int$[0:2]
...
```

If I use a queue instead (`x[$]`), then I get:
```
int[$]
```

So their return types are not the same and both should be `int[$]`. Do my thoughts make sense?

wsnyder, 2024-04-16:
A queue can be empty, so min needs to be able to represent that, unlike a foxes array which always has a min as zero-size is not legal.  Try your examples on other simulators.

goekce, 2024-04-16:
I think there is a misunderstanding. The LRM states that the return value of `min` *must* be a queue regardless of the datatype if I did not miss something.

You mention that *a fixed array has always a min as zero-size*. I don't understand this. Can you give an example?

> Try your examples on other simulators.

I tried the following on Edaplayground:

```sv
module m;
  int unsigned x[3] = '{0, 1, 2};
  initial begin
    $display(x.min);
    $display($typename(x.min));
    $finish;
  end
endmodule
```

#### Rivierapro:

```
# KERNEL: '{0}
# KERNEL: int unsigned$[$]
```

#### Questa

```
# 0
# int$[]
```

#### Synopsys VCS

```
'{'h0} 
int unsigned$[$]
```

wsnyder, 2024-04-27:
You're correct. Anyhow Verilator correctly returns a queue for unpacked arrays also.

goekce, 2024-04-28:
Should we revert back the original title for the issue

wsnyder, 2024-04-28:
Verilator returns correct values.  Saying `$display(x.min)` IEEE says is illegal use `$display("%p", x.min).

goekce, 2024-04-28:
I am confused. You first said that I am correct, but now you say something different :/ 

1. So you mean that displaying a queue without `%p` is illegal? Why does none of the simulators issue an error then? Or where does LRM state this? I could not find any info about this in *21.2.1.1 Format specifications*.

2. Verilator is still not correct about the return value: The return value of `min` *must* be a queue. Is this another issue?

wsnyder, 2024-04-28:
I stated earlier that min() returns in some cases a number, without trying it on Verilator.  You said it's always a queue return. You were correct.  And, Verilator *does* return a queue currently.  Using your example with the IEEE-required `display %p`:

```
'{'h0}
int$[0:2]

```

wsnyder, 2024-04-28:
IEEE 1800-2023 21.2.1.1: "An expression argument of any other unpacked data type that has no corresponding format specification shall be illegal.

I realized this as there is a commercial simulator that gave an error on your example.

goekce, 2024-05-01:
Oh, I did not notice that. So the only unpacked datatypes that may be displayed without any format specifiers are strings and array of bytes (I ask myself why only bytes). 

1. So both of the following examples should lead to errors:

```sv
module array_min;
  int unsigned array[3] = {0, 1, 2};
  initial begin
    $display(array.min);
    //$display(array.min[0]);  // Leads to compiler error, but this is another issue probably
    $display($typename(array.min));

    $finish;
  end
endmodule
```

Output:

```
3874012656
int$[0:2]
```
---

```sv
module queue_min;
  int unsigned queue[$] = '{0, 1, 2};
  initial begin
    $display(queue.min);
    $display(queue.min[0]);
    $display($typename(queue.min));

    $finish;
  end
endmodule
```

Output:

```
'{'h0} 
         0
int[$]
```

---

2. Moreover the return type of  `array.min` does not seem to be correct. It just shows the type of `array`. Probably this is  another issue. Should I open a new issue for this?

wsnyder, 2024-05-01:
1 & 2 are the same "new" issue that $typename(x.min) currently prints $typename(x).  If you assign to a queue both work correctly. Yes please file a new issue on that.

```
module array_min;
   int unsigned array[3] = {0, 1, 2};
   int unsigned queue[$] = '{0, 1, 2};
   int unsigned q[$];
   initial begin
      $display($typename(array));
      $display($typename(array.min));
      q = array.min;
      $display("%p is a %s", q, $typename(q));

      $display;
      $display($typename(queue));
      $display($typename(queue.min));
      q = queue.min;
      $display("%p is a %s", q, $typename(q));
      $write("*-* All Finished *-*\n");
      $finish;
  end
endmodule
```

```
int$[0:2]
int$[0:2]
'{'h0}  is a int[$]

int[$]
int[$]
'{'h0}  is a int[$]
```

wsnyder, 2024-05-01:
Actually the type problem is more similar to your initial report - the %p thing was just found trying to run it on other simulators.  I'll create another bug for %p and use this for the typename.

closed 2024-05-02 by wsnyder