verilator/verilator #5050: Internal Error: Variable inlining should make this impossible with v5.024

wltr, 2024-04-15:
When upgrading our code base to v5.024, we encountered the following error.
```
%Error: Internal Error: t/t_module_output_error.v:3:10: ../V3DfgPasses.cpp:314: Variable inlining should make this impossible
```
Running on Ubuntu 18.04.3 LTS. This can be reproduced using the test below.
```
// t_module_output_error_2.v
module t_module_output_error_2(input in, output out);
  assign out = in;
endmodule

// t_module_output_error.v
module t_module_output_error(input clk, output out);
  logic one = '1;
  t_module_output_error_2 dut_inst(.in(one), .out(out));
endmodule

// t_module_output_error.pl
#!/usr/bin/env perl
if (!$::Driver) { use FindBin; exec("$FindBin::Bin/bootstrap.pl", @ARGV, $0); die; }
# DESCRIPTION: Verilator: Verilog Test driver/expect definition
#
# Copyright 2021 by Wilson Snyder. This program is free software; you
# can redistribute it and/or modify it under the terms of either the GNU
# Lesser General Public License Version 3 or the Perl Artistic License
# Version 2.0.
# SPDX-License-Identifier: LGPL-3.0-only OR Artistic-2.0

scenarios(simulator => 1);

compile(
    verilator_flags2 => ["--sc t_module_output_error_2.v"]
    );

execute(
    );

ok(1);
1;
```

wsnyder, 2024-04-27:
@gezalore ?

gezalore, 2024-04-27:
Thanks for the ping. I will fix this. FWIW maybe reconsider #4417, which would eliminate a lot of these special cases.

closed 2024-04-27 by gezalore

gezalore, 2024-04-27:
Thanks for the nice test case @wltr This is now fixed on master.

PR #5076 (2024-04-27): Fix DFG assertion with SystemC
Fixes #5050
commit 27b7e702 (2024-04-27): Fix DFG assertion with SystemC (#5076)

Fixes #5050