verilator/verilator #5051: Verilation makefile should not use `$(info Archive ...)`.

goekce, 2024-04-16:
In my verification flow I use assertions and thus I pay attention that the terminal is not cluttered. My current solution is to execute in my makefile the following:

```sh
verilator -MAKEFLAGS '--quiet' --quiet --binary m.sv && obj_dir/Vm +verilator+quiet
```

Verilator uses make for every verilation process. So I use `--quiet` to suppress the output of make. However `--quiet` only suppresses the string of the executed commands (like `@`) and it does not suppress `$(info ...)` or the output of an `echo`. So the following line cannot be suppressed (without a `grep -v` etc):

https://github.com/verilator/verilator/blob/1315aa31ed1b980b05778f5a0a1f3ff9718be680/include/verilated.mk.in#L228

The compiler commands during verilation do not make use of `$(info ..)` and they can be silenced using `-MAKEFLAGS '--quiet'`.

My proposal is that we remove this line. What do you think?

wsnyder, 2024-04-16:
OK, please submit a pull to remove it, thanks.

goekce, 2024-04-20:
After [discussing with an experienced `make` user](https://savannah.gnu.org/bugs/index.php?65600) I found out:

Using `.SILENT` and summarizing `$(info _summary_)` is an alternative method to have a less verbose but informative output during make. If less verbose output is desired, then $(info) can be suppressed by using `>/dev/null` or making it configurable in the makefile. Actually this way of suppressing the output already doable without #5059 using:

 ```sh
verilator --cc m.sv
make -C obj_dir -f Vm.mk > /dev/null
```

Presumably most users of Verilator use makefiles for Verilation so the code above should not be a big deal. However I still believe the `$(info ar ..)` does not additional value in the current structure of the makefile.

PR #5059 (2024-04-20): Remove $(info) which cannot be silenced by `-MAKEFLAGS '--quiet'` (#5051)
Closes #5051 .
closed 2024-04-20 by wsnyder