verilator/verilator #5052: No warning/error if a signal is driven by both concurrent and continuous assignment

goekce, 2024-04-16:
### Example

```sv
module m;
  logic clk = 1;
  assign #1 clk = !clk;

  initial begin
    $dumpfile("dump.vcd");
    $dumpvars;
    #10 $finish;
  end
endmodule
```

I tried on Vivado, Questa and Rivierapro and they error out. For example RivieraPro:

```
Variable "clk" cannot be driven by both the procedural and continuous assignment statements. Recompile using "-err VCP2980 W1" to suppress this error
```

### Expected result

Verilator should at least flag a warning.

### Version

5.024 on Archlinux

### Related

Could be related: https://github.com/verilator/verilator/issues/1369

wsnyder, 2024-04-27:
#1369 is similar but not obviously the same.  Might you be willing to submit a pull to add this warning?

goekce, 2024-04-27:
>  Might you be willing to submit a pull to add this warning?

I would like to, however based on my negative experience with another issue (3875) I have the impression that I cannot fix this in reasonable time.