verilator/verilator #5054: timescale option not work (version5.025)

dinggxi, 2024-04-18:
Hello, I'm new to Verilator and I'm working on Ubuntu 20.04.6 LTS with Verilator version 5.025. I'm trying to generate a VCD file, but I'm encountering an issue with the timescale(but the waveform is correct). Despite adding the option ```--timescale 1ns/1ns``` or including `timescale 1ns/1ns in my Verilog file header, the timescale remains at 1s. The option seems not work?
Here is my testbench to generate a VCD file:
```
#include <verilated.h>
#include <verilated_vcd_c.h>
#include "Vtop_tb.h"
 
int main(int argc, char** argv){
	
	VerilatedContext* contextp = new VerilatedContext;
	VerilatedVcdC* trace = new VerilatedVcdC;
	Vtop_tb* top = new Vtop_tb;

	contextp->traceEverOn(true);
	top->trace(trace,3);	
	trace->open("Vtop_tb.vcd");
	
	while (!contextp->gotFinish()){
		top->eval();
		trace->dump(contextp->time());
		contextp->timeInc(1);
	}

	trace->close();
	delete top;
	return 0;
}
```
I've also tried modifying sim_main.cpp in the /example/make_tracing_c file with the same logic as the testbench code mentioned earlier to generate a VCD file. However, I'm encountering the same issue with the timescale.
```
 /* <...> */
#include <verilated.h>
#include <verilated_vcd_c.h>
 /* <...> */
int main(int argc, char** argv) {
	/* <...> */
	contextp->traceEverOn(true);
	VerilatedVcdC* m_trace = new VerilatedVcdC;
	/* <...> */
	const std::unique_ptr<Vtop> top{new Vtop{contextp.get(), "TOP"}};
	top->trace(m_trace, 0);
	m_trace->open("Vtop.vcd");
	/* <...> */
	while (!contextp->gotFinish()) {
		/* <...> */
		VL_PRINTF("[%" PRId64 "] clk=%x rstl=%x iquad=%" PRIx64 " -> oquad=%" PRIx64
		               " owide=%x_%08x_%08x\n",
		                contextp->time(), top->clk, top->reset_l, top->in_quad, top->out_quad,
		                top->out_wide[2], top->out_wide[1], top->out_wide[0]);
		m_trace->dump(contextp->time());
	}
    m_trace->close();
    top->final();
    /* <...> */
```
the following command I used in /example/make_tracing_c to simulate:
```
$verilator --timescale 1ns/1ns --timing --trace --cc top.v --exe sim_main.cpp
```
```
$make -C obj_dir -f Vtop.mk
```
```
$ ./obj_dir/Vtop
[1] Model running...

[1] clk=1 rstl=1 iquad=1234 -> oquad=1235 owide=3_22222222_11111112
[2] clk=0 rstl=0 iquad=1246 -> oquad=0 owide=0_00000000_00000000
[3] clk=1 rstl=0 iquad=1246 -> oquad=0 owide=0_00000000_00000000
[4] clk=0 rstl=0 iquad=1258 -> oquad=0 owide=0_00000000_00000000
[5] clk=1 rstl=0 iquad=1258 -> oquad=0 owide=0_00000000_00000000
[6] clk=0 rstl=0 iquad=126a -> oquad=0 owide=0_00000000_00000000
[7] clk=1 rstl=0 iquad=126a -> oquad=0 owide=0_00000000_00000000
[8] clk=0 rstl=0 iquad=127c -> oquad=0 owide=0_00000000_00000000
[9] clk=1 rstl=0 iquad=127c -> oquad=0 owide=0_00000000_00000000
[10] clk=0 rstl=1 iquad=128e -> oquad=128f owide=3_22222222_11111112
[11] clk=1 rstl=1 iquad=128e -> oquad=128f owide=3_22222222_11111112
[12] clk=0 rstl=1 iquad=12a0 -> oquad=12a1 owide=3_22222222_11111112
[13] clk=1 rstl=1 iquad=12a0 -> oquad=12a1 owide=3_22222222_11111112
[14] clk=0 rstl=1 iquad=12b2 -> oquad=12b3 owide=3_22222222_11111112
[15] clk=1 rstl=1 iquad=12b2 -> oquad=12b3 owide=3_22222222_11111112
[16] clk=0 rstl=1 iquad=12c4 -> oquad=12c5 owide=3_22222222_11111112
*-* All Finished *-*
- sub.v:29: Verilog $finish
[17] clk=1 rstl=1 iquad=12c4 -> oquad=12c5 owide=3_22222222_11111112
- S i m u l a t i o n   R e p o r t: Verilator 5.025 devel
- Verilator: $finish at 17ns; walltime 0.001 s; speed 0.000 s/s
- Verilator: cpu 0.000 s on 1 threads; alloced 249 MB
```
The head 2 lines of the produced Vtop.vcd file, there is a problem with the time scale:
```
$version Generated by VerilatedVcd $end
$timescale 1s $end
```
If there are any issues with my testbench or the command line, please assist me in correcting them. Thank you for all your assistance.

wsnyder, 2024-04-18:
You need to construct the model before you construct the trace, so that the trace knows the proper timescale (which comes from the model).  This is confusing and I'll try to clean it up.

That is swap to this:

```
	Vtop_tb* top = new Vtop_tb;
	VerilatedVcdC* trace = new VerilatedVcdC;
```

dinggxi, 2024-04-19:
It works. Thank for your help!

closed 2024-04-19 by dinggxi