verilator/verilator #5060: Compile error when using module variable in class defined under module

tudortimi, 2024-04-21:
I have the following code:

```
module test;

    bit some_bit;

    class some_class;

        task do_stuff();
            some_bit = 1;
        endtask

    endclass

endmodul
```

Verilator is giving me the following error:

```
%Error: Internal Error: test.sv:8:13: ../V3Scope.cpp:73: Can't locate varref scope
                                    : ... note: In instance 'test.some_class'
    8 |             some_bit = 1;
      |             ^~~~~~~~
```

I'm using Verilator 5.024. I ran with `verilator test.sv --binary`.

wsnyder, 2024-04-22:
Might you prepare a pull to fix this?

wsnyder, 2024-04-24:
Not sure, V3Scope probably isn't correctly tracking classes underneath modules, given both classes and modules are subclasses of AstNodeModules.