verilator/verilator #5061: Fusing macro argument easts up whitespace

tudortimi, 2024-04-21:
I have the following code:

```
class some_class;
endclass

`define some_macro(NAME) \
    class extension_for_``NAME`` extends some_class; \
    endclass \

`some_macro(foo)
```

I get the following error:

```
%Error: test.sv:8:37: syntax error, unexpected TYPE-IDENTIFIER, expecting ';'
    8 |     class  extension_for_fooextends some_class; 
      |                   
```

You can see that Verilator is not leaving a ` ` between `foo` and `extends`.

I'm using Verilator 5.024 and running with `verilator test.sv --binary`.

wsnyder, 2024-04-22:
Surprising this was missed.

Might you be able to look at the V3PreProc code to see what's causing this and prepare a pull request?  The t_preproc.v test should be updated to have this case in it.

tudortimi, 2024-04-22:
This sounds like it would be simple enough. Sure, assign it to me.

tudortimi, 2024-04-29:
I had a look at the LRM and the trailing ` `` ` aren't needed for fusing `class_for_` and `NAME`. In this case, ` extension_for_``NAME ` would be enough.

I'm not really 100% sure what ` NAME`` something ` should produce in this case. The LRM states:

> A `` delimits lexical tokens without introducing white space [...].

Is ` ` considered a lexical token? Should it simply expand to the value of `NAME` followed by ` something`, as if simply defining `NAME something`? (Other tools seem to do this.)

tudortimi, 2024-04-29:
Assuming we want to fuse like in the above post, like all the other tools do, I had a look at the code and think I kinda got it.

I have the following macro in the test:

```
`define with_space_before_suffix(f) f`` suffix_after_space
`with_space_before_suffix(arg)
```

I see the preprocessor recognizing the ` arg`` ` part:

[image]
[image]

After `arg` gets put onto the stack and the state is set to `JOIN`, it hits the whitespace:

[image]
[image]

This line causes the space to be emitted and keeps the state as `JOIN`. (I'm guessing it then somehow jumps into the `GETFETC` part, which takes care of outputting the space, but didn't really follow how it gets there.) This explains why the expansion starts with ` arg` (with space) and not `arg` (no space).

The next token is `suffix_after_space`, which also gets pushed to the stack as the RHS of the `JOIN` (the state left over):

[image]

This is because it still thinks it's in the `JOIN` that the ` ` should have actually ended.

I don't really know what code is needed to fix this, but I'm guessing there's some extra case needed to handle `WHITE` when in `JOIN` here:

[image]

wsnyder, 2024-04-30:
Your analysis seems correct, including likely some new text is needed to handle WHITE when in JOIN in your post above.

wsnyder, 2024-06-12:
@tudortimi, ping, do you think you'll be making a pull for this?

tudortimi, 2024-06-13:
I got kind of stuck with it when I tried. I tried multiple things and couldn't really get it to work.

I added the following test:

```
diff --git a/test_regress/t/t_preproc.v b/test_regress/t/t_preproc.v
index d07a42994..0848b83cc 100644
--- a/test_regress/t/t_preproc.v
+++ b/test_regress/t/t_preproc.v
@@ -83,6 +83,9 @@ $display(`msg(left side, right side))
 `define foo(f) f``_suffix
 `foo(bar)  more

+`define with_space_before_suffix(f) f`` suffix_after_space
+`with_space_before_suffix(arg)
+
 `define zap(which)   \
        $c("Zap(\"",which,"\");");
 `zap(bug1);

```

I tried multiple things, but I still got the following in my `git diff` from my last unsuccesfull attempt:

```
@@ -989,7 +989,7 @@ int V3PreProcImp::getStateToken() {
         int tok = getRawToken();

         // Most states emit white space and comments between tokens. (Unless collecting a string)
-        if (tok == VP_WHITE && state() != ps_STRIFY) return tok;
+        if (tok == VP_WHITE && state() != ps_STRIFY && state() != ps_JOIN) return tok;
         if (tok == VP_BACKQUOTE && state() != ps_STRIFY) tok = VP_TEXT;
         if (tok == VP_COMMENT) {
             if (!m_off) {
@@ -1430,6 +1430,7 @@ int V3PreProcImp::getStateToken() {
                 const string lhs = m_joinStack.top();
                 m_joinStack.pop();
                 unputString(lhs);
+                unputString(" ");
                 statePop();
                 goto next_tok;
             } else {
```

It was pretty difficult for me to wrap my head around the handling of states and tokens. I for sure need help with this. At the same time, I'm not sure whether it's going to be easier for you to give me a hint or to just solve it yourself, as I feel there's something more tricky to do here.

wsnyder, 2024-06-15:
Your patch was almost complete so very appreciated, fixed!

closed 2024-06-15 by wsnyder