verilator/verilator #5062: Probable bug in force behavior

jackkoenig, 2024-04-22:
> Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
>
> What 'verilator' command line do we use to run your example?

I have a reproducible test case in this repo (https://github.com/jackkoenig/verilator-force-issue),

> What 'verilator --version' are you using?  Did you try it with the git master version?

Code works fine in Verilator v5.022 and before, but is present in Verilator v5.024 and current head of master (a9664b8d1).
I bisected the issue to be introduced in commit https://github.com/verilator/verilator/commit/5e1fc6e24d9c2706d9871de9bec25cebf2a95ac7

> What OS and distribution are you using?

macOS Ventura 13.5, verilator built from source

I'll paste the contents of the README here:

# Verilator Force Issue

**Disclaimer**: Apologies that the C++ wrapping the Verilator simulation is a more complex than it needs to be.
I believe the problem is understandable from just the Verilog so I figured I wouldn't bother minimizing that infrastructure any further.

# Introduction

This is a reproducible test case for a probable bug in Verilator v5.024

You can run the test with `make replay` (or just `make`).
It will default to using the `verilator` on your `PATH`, but you can set the `VERILATOR` environment variable to point to a different `verilator` binary.

Correct operation, e.g. using Verilator v5.022 results in the following output (and an exit value of `0`):

```
r ready
k ack
b 00000040 000000000000000A
k ack
[ 0]: dut.r = 0x002a, dut.out = 0x002a
[ 1]: dut.r = 0x002a, dut.out = 0x002a
[ 2]: dut.r = 0xdead, dut.out = 0xbeef
[ 3]: dut.r = 0xdead, dut.out = 0xbeef
b 00000040 0000000000000003
grep 'Verilog $finish' /Users/koenig/work/verilator-testcase-simple/simulation-log.txt
- primary-sources/ProbeSpec.sv:26: Verilog $finish
```

Incorrect operation using Verilator v5.024 results in the following output (and a non-zero exit value):

```
r ready
k ack
b 00000040 000000000000000A
k ack
[ 0]: dut.r = 0x002a, dut.out = 0x002a
[ 1]: dut.r = 0x002a, dut.out = 0x002a
[ 2]: dut.r = 0xbeef, dut.out = 0xbeef
/bin/sh: line 1: 19207 Done                    cat /Users/koenig/work/verilator-testcase-simple/execution-script.txt
     19208                       | sed -n 's/^[0-9]*> \(.*\)/\1/p'
     19209 Abort trap: 6           | SVSIM_SIMULATION_LOG=/Users/koenig/work/verilator-testcase-simple/simulation-log.txt SVSIM_SIMULATION_TRACE=/Users/koenig/work/verilator-testcase-simple/trace /Users/koenig/work/verilator-testcase-simple/simulation
grep 'Verilog $finish' /Users/koenig/work/verilator-testcase-simple/simulation-log.txt
make: *** [replay] Error 1
```

## Diagnosis

The issue is with forcing both a register and an output port driven by that register.
The target code in question:
```verilog
// From primary-sources/Top.sv
module Top(
  input         clock,
  output [15:0] out
);
  reg [15:0] r;
  always @(posedge clock)
    r <= 16'h2A;
  assign out = r;
endmodule
```

These values are forced by a module above:
```verilog
// From primary-sources/ProbeSpec.sv

force ProbeSpec.dut.r = 16'hdead;
// If you comment out the below line, the $fatal below does not occur
force ProbeSpec.dut.out = 16'hbeef;
```

When both `Top.r` and `Top.out` are forced to different values, `Top.r` should get the value it is forced to, but instead, it is getting the value forced to `Top.out` as illustrated in the output above.

I bisected the issue to be introduced in commit https://github.com/verilator/verilator/commit/5e1fc6e24d9c2706d9871de9bec25cebf2a95ac7

seldridge, 2024-04-23:
My high level understanding of that commit is that it will add intermediate variables for situations where there are multiple sinks of a vertex: https://github.com/verilator/verilator/commit/5e1fc6e24d9c2706d9871de9bec25cebf2a95ac7#diff-b97010442f078f1560701a463b217dcdb608556f4e3757d896256fe1e440034cR83 However, here the variable and the net cannot be combined even though the variable only has one sink (the net). However, this breaks force semantics because a force of a net shouldn't force it's driving variable. (While it is correct for a force of a net to travel "backwards" and force a driving net.)

wsnyder, 2024-04-27:
A patch to fix this would be appreciated.

seldridge, 2024-04-29:
More archaeology, it looks like there was a related bug which was partially fixed with: https://github.com/verilator/verilator/pull/4942 The fix may be a similar thing where `isForceable` is queried to short circuit the optimization of this pass.

kozdra, 2024-04-30:
The reproducer code works correctly with `-fno-inline` as a workaround.
The source can also be made to work by adding a `/* verilator public */` attribute to the output variable being forced.

A full fix requires adding something like the other `/* verilator forceable */` attribute to the variable being forced during some new force assignments analysis phase before V3Inline elliminates the output variable. @wsnyder WDYT?

wsnyder, 2024-05-01:
@gezalore thoughts on the DFA implications?  It's interesting that the inliner worked in this case previously.

gezalore, 2024-05-01:
Signals that are forced from source via a `force` statement should be marked as having an external reference in DFG (there exists an AstNodeVarRef in some code DFG cannot represent - namely, under the AstAssignForce), and it should leave alone any such references. If theory fails and `-fno-dfg` fixes it I'm happy to look further at a small stand alone testcase. Can you please checl/provide this @jackkoenig.

kozdra, 2024-05-02:
I managed to reduce the code down to ~50 lines. Running the test fails by itself, but adding either `-fno-inline` or `-fno-dfg` fixes it, and so does annotating the problematic variable as public.
```sv
`define stop $stop
`define checkh(gotv,expv) do if ((gotv) !== (expv)) begin $write("%%Error: %s:%0d:  got='h%x exp='h%x\n", `__FILE__,`__LINE__, (gotv), (expv)); `stop; end while(0)

module sub(
   // Outputs
   out,
   // Inputs
   clk
   );
   // verilator inline_module

   output [3:0] out /* <-- this variable has to be marked as having external refs */;
   input clk;

   reg [3:0] r;

   always @ (posedge clk)
      r <= 4'h1;
   assign out = r;
endmodule

module t(/*AUTOARG*/
   // Inputs
   clk
   );

   input clk;
   reg [3:0] unused;

   sub sub1(unused, clk);

   integer cyc = 0;

   always @ (posedge clk) begin
      cyc <= cyc + 1;
      if (cyc == 1) begin
         `checkh(sub1.r, 4'h1);
         `checkh(sub1.out, 4'h1);
      end
      else if (cyc == 2) begin
         force sub1.r = 4'h2;
         force sub1.out = 4'h3;
      end
      else if (cyc == 3) begin
         `checkh(sub1.r, 4'h2);
         `checkh(sub1.out, 4'h3);
      end
      //
      else if (cyc == 99) begin
         $write("*-* All Finished *-*\n");
         $finish;
      end
   end

endmodule
```

gezalore, 2024-05-02:
Thanks for the test case.

> My feeling is that DFG deals improperly with assign aliases.

You are quite right that this has to do with aliases (AstAssignAlias). But I'm going to claim this is a bug in the Inliner, not DFG.

The DFG patch that introduced this just made DFG more aggressive so this bug now shows more easily.

Here is my understanding. Given:

```verilog
module top(input wire a, output wire b);
   sub subi(a, b);
endmodule

module sub(input wire i, output wire o);
  assign o = i;
endmodule
```

When we inline `subi` into `top`, it will create 2 aliases for the port `i` and `o`:
```
1:2: ASSIGNALIAS 0x555556c1fcc0 {d7ax} @dt=0x555556c1c840@(G/w1)
1:2:1: VARREF 0x555556bf8dd0 {d4an} @dt=0x555556c1c840@(G/w1)  top__DOT__a [RV] <- VAR 0x555556c28d80 {d3ax} @dt=0x555556c1c840@(G/w1)  top__DOT__a [VSTATIC]  WIRE
1:2:2: VARREF 0x555556bf8d00 {d7ax} @dt=0x555556c1c840@(G/w1)  top__DOT__subi__DOT__i [LV] => VAR 0x555556c29200 {d7ax} @dt=0x555556c1c840@(G/w1)  top__DOT__subi__DOT__i [VSTATIC]  WIRE
1:2: ASSIGNALIAS 0x555556c1fc20 {d7bm} @dt=0x555556c1c840@(G/w1)
1:2:1: VARREF 0x555556bf84e0 {d4aq} @dt=0x555556c1c840@(G/w1)  top__DOT__b [RV] <- VAR 0x555556c28ea0 {d3bm} @dt=0x555556c1c840@(G/w1)  top__DOT__b [VSTATIC]  WIRE
1:2:2: VARREF 0x555556bf85b0 {d7bm} @dt=0x555556c1c840@(G/w1)  top__DOT__subi__DOT__o [LV] => VAR 0x555556c29320 {d7bm} @dt=0x555556c1c840@(G/w1)  top__DOT__subi__DOT__o [VSTATIC]  WIRE
```

At a later pass, all `AstAssignAlias` LHS is replaced with the RHS, so eventually the inlined sub-module ports `top.subi.i` and `top.subi.o` are completely eliminated and references replaced with `top.a` and `top.b` respectively.

This replacement happens in the `force` statement as well, so in your example `force top.out = ...` actually behaves as if you typed `force unused = ...`.

However, DFG does not know that this replacement will happen later, and nor does it know that `top.subi.o` will eventually be treated as `top.unused`. With the recent DFG change, it can also deduce after inlining that `top.unused` is equivalent to `top.subi.r`, but it doesn't see that `top.unused` is forceable, as the `force` statement is referencing `top.subi.o`, so it goes ahead and replaces `top.unused` with `top.subi.r` (includign in the alias). We then later replace `top.subi.o` with `top.subi.r` due to the alias.

> EDIT: another workaround, more related to DFG, is that /* verilator forceable */ can be added to the reg [3:0] unused variable

That's why this workaround fixes it. It tells DFG that `top.unused` is forceable, so it doesn't replace it with `top.subi.r`, even though `top.unused` is not really forceable (it's `top.subi.o` that is really forceable).

In fact, if I replace `force sub.o = 3` with `force unused = 3` in your test, it will pass, as in reality that is forcing `top.subi.o` (which is what your test checks). This is completely wrong though, forcing `unused` should not force it's driver `top.subi.o`.

I could fix this in DFG, by disabling optimizations for signals that are used as the RHS of an AstAssignAlias, but that would have the unfortunate side-effect of unnecessarily reducing optimization on *all* port signals, even those that are not forceable (this can be worth ~10% speed in some cases). Also I think this would not fix other issues, e.g, when  you would try to force `top.subi.clk` (an input), it would also force `top.clk` (what drives that input), which is also wrong and has nothing to do with DFG.

I believe the proper fix is for V3Inline not to create aliases for forceable signals, just like it doesn't create AstAssignAlias (but use AstAssignW instead) for some public signals. See here, it's essentially the same reason explained in the comment:
https://github.com/verilator/verilator/blob/8044833c74e99b2d8bc13026acae662ee5e50374/src/V3Inline.cpp#L547-L552

@wsnyder suggestions?

wsnyder, 2024-05-03:
@gezalore your analysis sounds correct.  There's no current AstVar flag that indicates a var (might) be forced, so either V3Inline needs to temporarily figure that out from the recursion in InlineMarkVisitor when it sees VarRef/VarXRef's under a AstAssignForce, or some earlier stage e.g. Link needs to similarly set a new flag in AstVar.  Then that information will suppress the making of AssignAlias.

gezalore, 2024-05-10:
This should now be fixed on master, but let me know if you still have issues.

closed 2024-05-10 by gezalore

jackkoenig, 2024-05-13:
I can confirm this is fixed for us, thank you @gezalore!

PR #5105 (2024-05-10): Do not create aliases for forced port signals
+ don't remove forced signals in V3Const and Dfg

Fixes #5062
commit cf111d2e (2024-05-10): Do not create aliases for forced port signals (#5105)

+ don't remove forced signals in V3Const and Dfg

Fixes #5062