verilator/verilator #5066: Errors when using a nested SytemVerilog interface as a port connection in a module

sterin, 2024-04-25:
First, an internal error is reported when a nested interface is used as a module port connection. The following code

```SystemVerilog
interface A;
    logic x;
endinterface

interface B;
    A a();
endinterface

module M(A a);
endmodule

module top;
    B b();
    M m(b.a);
endmodule
```

results in the following error:

```
%Error: Internal Error: bug1.sv:14:11: ../V3Scope.cpp:73: Can't locate varref scope
   14 |     M m(b.a);
      |           ^
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

Second case, this time the nested interface is parameterized. The following code

```SystemVerilog
interface A #(WIDTH);
    logic [WIDTH-1:0] x;
endinterface

interface B;
    A #(32) a();
endinterface

module M(A a);
endmodule

module top;
    B b();
    M m(b.a);
endmodule
```

results in the following error:

```
%Error: bug2.sv:1:15: Parameter without initial value is never given value (IEEE 1800-2023 6.20.1): 'WIDTH'
                    : ... note: In instance 'top.m'
    1 | interface A #(WIDTH);
      |               ^~~~~
%Error: bug2.sv:9:10: Parent instance's interface is not found: 'A'
                    : ... note: In instance 'top.m'
    9 | module M(A a);
      |          ^
%Error: Exiting due to 2 error(s)
```

Both examples work in VCS/Questa/Xcelium/Reviera Pro (tested on EDA Playground), and also in Vivado.

> Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

I'll open a pull request with the tests in test_regress format, and update this issue accordingly.

> What 'verilator' command line do we use to run your example?

```
verilator --timing --trace-fst --top-module top --binary bug.sv
```

> What 'verilator --version' are you using?  Did you try it with the git master version?

5.018, 5.024, and git master as of today.

> What OS and distribution are you using?

Ubuntu 22.04 (WSL).

> May we assist you in trying to fix this in Verilator yourself?

If this look like someone who is a Verilator beginner could reasonably handle.

wsnyder, 2024-04-25:
Thanks for the good test case.  Interfaces are already fairly complicated and this is doubly so, thus it may be a while before this gets supported/fixed, if it's important to you a pull to fix it would be appreciated.

cachanova, 2026-02-02:
Submitted a fix in PR #6986.